RobotLocomotion/drake #4091: Move RBT parsing files into `drake/multibody/parsers`

amcastro-tri, 2016-11-09:
Once #4058 is merged, move parsing files into their own directory.
What do you think @liangfok?

jwnimmer-tri, 2016-11-09:
Yes in general the pile of files in `multibody` need further reorganization.  But given the inappropriate class designs (circular dependencies) this is not as simple as just mv'ing some files.

amcastro-tri, 2016-11-09:
I agree but, would this be a step further towards this reorganization? I think there are no circular dependencies here that would not allow me to do this refactoring.

jwnimmer-tri, 2016-11-09:
There is a circle now, yes.  The RBT includes the parsers which include the RBT.

liangfok, 2016-11-09:
One concern I have is the parsers may emit more information than what's stored in `RigidBodyTree`. For example, SDF contains light information, which doesn't seem to belong in a `RigidBodyTree`:

http://sdformat.org/spec?ver=1.6&elem=light

How about let's put the parsers in `drake-distro/drake/parsers`?

jwnimmer-tri, 2016-11-09:
@liangfok The mandate of `multibody` will be wide and deep.  Not only the kinematic tree, but also the visual scene (including lights), actuators, sensors, etc.  I think more or less we should keep everything that's already in there still in there, until we have a positively-identified need to elevate it.

amcastro-tri, 2016-11-09:
Humm.... isn't all the parsing code now in `drake/multibody` specific to RBT? in which case moving to `multibody/parsers` would be fine?

liangfok, 2016-11-09:
Yes. I was just thinking in the future where we will have parsers for things like RNDF, XODR, etc. Will those be in different `parsers/` directories?

jwnimmer-tri, 2016-11-09:
Umm... yes?

liangfok, 2016-11-09:
Cool. SGTM...until a single format spans multiple `parsers/` directories.

jwnimmer-tri, 2016-11-09:
If there exists an input format that merges both SDF and RNDF, I will burn it with fire, rather than write a parser for it.

liangfok, 2016-11-23:
Consider renaming `parser_urdf` and `parser_sdf` to be `urdf_parser` and `sdf_parser`, respectively, along with their relocation.

liangfok, 2016-11-23:
How about let's place all parsers in `drake-distro/parsers`?

amcastro-tri, 2016-11-23:
The parsers no in `drake/multibody` are very RBT specific. I would therefore put those in `drake/multibody/parsers`.

liangfok, 2016-11-23:
Sorry, I meant to say `drake-distro/drake/parsers`.

OK we can put them in `drake-distro/drake/multibody/parsers` for now, and then move them into a more general location in case they eventually become not multibody specific.

PR #4482 (2016-12-14): Moves SDF and URDF Parsers Into libdrakeMultibodyParsers
Resolves #4091.

---
This change is [[image]](https://reviewable.io/reviews/robotlocomotion/drake/4482)
closed 2016-12-14 by liangfok