RobotLocomotion/drake #4112: Decouple DrakeVisualizer from the RigidBodyTree

liangfok, 2016-11-10:
In preparation for the existence of `GeometryWorld`, it makes sense to remove `DrakeVisualizer`'s dependence on `RigidBodyTree`.

jwnimmer-tri, 2016-11-10:
To be clear: we should retain a publisher System that uses the RBT, like we have now.  But we _also_ want to have a System that publishes to drake-vis that _doesn't_ need the Tree.  So two separate classes (that likely share a lot of code).  This is a stop-gap until GeometryWorld comes online and supplants (refactors) this new System.

SeanCurtis-TRI, 2016-11-10:
Rather than me digging through the code, can someone summarize what the current dependency is?

jwnimmer-tri, 2016-11-10:
We want a System very much like this one:
https://github.com/RobotLocomotion/drake/blob/ddbde0448fa38e271a08706e2457bb2e61fff351/drake/multibody/rigid_body_plant/drake_visualizer.h#L58

But without mentioning `RigidBodyTree` (and thus, with a different input port type).

liangfok, 2016-11-10:
`DrakeVisualizer` publishes two types of messages:

1. `lcmt_viewer_load_robot` - upon init
2. `lcmt_viewer_draw` - each update

The `lcmt_viewer_load_robot` contains visual geometry information, which it currently obtains from the `RigidBodyTree`. We want it to be able to get this geometry info from other things, like `GeomtryWorld`.

The `lcmt_viewer_draw` message does not require `RigidBodyTree`.

jwnimmer-tri, 2016-12-24:
I've renamed the issue to make it clear what the point is: do we want to stand up a temporary `lcmt_viewer` publisher that can operate without using RigidBodyTree, prior to GeometryWorld coming online, so that the `automotive` library and demos are easier to develop in the meantime?

rdeits, 2017-01-06:
BTW, this is exactly what I've done with https://github.com/rdeits/DrakeVisualizer.jl/ which wraps the interface to the visualizer and https://github.com/rdeits/RigidBodyTreeInspector.jl which uses that interface to draw rigid body tree robots. 

Also, FYI, there's a major redesign of the drake visualizer interface underway: https://github.com/RobotLocomotion/director/pull/420 with the goal of supporting more geometry types and more flexible communication between the viewer and client.

david-german-tri, 2017-01-06:
> I've renamed the issue to make it clear what the point is: do we want to stand up a temporary lcmt_viewer publisher that can operate without using RigidBodyTree, prior to GeometryWorld coming online, so that the automotive library and demos are easier to develop in the meantime?

Yes (in part because it's a useful prototyping step for GeometryWorld).  I'll work on it.

david-german-tri, 2017-01-06:
@rdeits Thanks for the pointers.  I agree, this ticket proposes to do something similar (in C++ System2 diagrams) to what you've done (in a separate Julia app).  I have a couple of comments that I'll add to RobotLocomotion/director#420.

david-german-tri, 2017-04-03:
This was fixed in https://github.com/RobotLocomotion/drake/pull/5476

closed 2017-04-03 by david-german-tri