RobotLocomotion/drake #4113: Proliferation of Shared Libraries in KUKA IIWA Example Needs Consolidation

liangfok, 2016-11-11:
# Problem Definition

There is a proliferation of libraries in `drake-distro/drake/examples/kuka_iiwa_arm/`:

* `drakeKukaIiwaArmCommon`
* `drakeKukaIiwaArm`
* `drakeKukaIiwaWorld` (pending #4107)

We should reconsider whether having so many libraries is reasonable. For example, maybe all three can be consolidated into one library after System 1.0 is removed.

naveenoid, 2016-11-11:
Very good point. I will consolidate as soon as #4107 lands.

naveenoid, 2016-11-28:
Status update : The result of #4107 can easily be merged with `drakeKukaIiwaArmCommon` which in turn can be renamed to `drakeKukaIiwaArm` once issue #3977 is resolved. That is currently blocked by the status of `kuka_plan_runner.cc and `kuka_lcm_visualizer`.

PR #6738 (2017-09-29): Purge CMake not used by MATLAB
Toward #3129. Closes #3342, closes #3426, closes #4113, closes #4525, closes #5356, and closes #5572.

----
This change is [[image]](https://reviewable.io/reviews/robotlocomotion/drake/6738)
closed 2017-09-29 by jamiesnape