RobotLocomotion/drake #4114: Need Diagram-level DoPublish() method

sherm1, 2016-11-11:
@RussTedrake asked me how to capture a trajectory during a simulation. I said, you just need to capture a copy of the Context at each simulation step, or at periodic intervals. You can do that with the Publish() method since it gets called with a const Context at specified sampling times. Russ said, but that only provides the local LeafSystem's subcontext. I said, no, capture it in the Diagram's Publish() ... oops.

Currently I don't think there is a way for a user to provide Diagram-level Publish() behavior. Am I missing something? Otherwise we should provide that so the full Context can be captured.

(A workaround is to call `Simulator::StepTo()` repeatedly for short periods and ask the Simulator for the Context after each interval.)

RussTedrake, 2016-11-11:
But I still don't think this adequately addresses #3734.  The request is that the simulator not only returns the data as evaluated at each time step (in a trajectory object), but that it returns the correct interpolation scheme given the integration method that was used.  E.g., if ode45 gives 4th order accuracy, then perhaps it returns a fourth order spline trajectory.

david-german-tri, 2016-11-11:
Yeah, I always assumed this would be implemented in general form in the Simulator, not in user-specific Diagram code.  However, if whole-Diagram Publish is the right thing to do, it would be easy enough to add hooks for that post #4031.

sherm1, 2016-11-11:
> But I still don't think this adequately addresses #3734. 

I agree. That's a bigger project. But it should be possible to publish whole-Context results from a Diagram; we didn't have a use case before.

PR #12213 (2019-11-05): Adds a monitor function for Simulator trajectories
Adds an optional monitor() function to a Simulator() that can be used to publish trajectories, check for termination, or report errors. Also adds a status return from Simulator::AdvanceTo() and related methods so that the monitor() results can be reported (backwards compatible if ignored).

The monitor() function is semantically identical to a top-level, per-step Publish event handler. It uses the pre-existing EventStatus class to report success, termination, or failure.

Resolves #12164
Resolves #4114

---
This change is [[image]](https://rev [... cut]
closed 2019-11-05 by sherm1