RobotLocomotion/drake #4121: quaternion_floating_joint.h gives up four digits of accuracy on forward / backward conversion

edrumwri, 2016-11-11:
This issue concerns QuaternionFloatingJoint::v2qdot() and QuaternionFloatingJoint::qdot2v(). Over various orientations, converting an angular velocity to a time derivative of a unit quaternion and back can lose between four and five digits of precision.

This code is also doing much far computation than is necessary. Multiplications between a 4x3 matrix and 3x1 / 4x1 vectors is all that is required.

When this issue is sorted, the zero tolerance in rigid_body_plant_test.cpp should be able to be decreased from 10^-10 to at most 10^-14. A proposed fix for this issue should succeed at that test.

Setting to high priority as this affects the accuracy of the simulation.

Assigning to @tkoolen, as I've been led to believe this was the original author.

sherm1, 2016-11-17:
In case this is useful, here is the near-trivial code for producing the v2qdot matrix N(q) for a quaternion q such that `qdot=N*v`:
```c++
/** Given a possibly unnormalized quaternion q, calculate the 4x3 matrix N which
maps angular velocity w to quaternion derivatives qdot. We expect the angular 
velocity in the parent frame, i.e. w==w_PB_P. We don't normalize, so N=|q|N' 
where N' is the normalized version. Cost is 7 flops. **/
static Mat43P calcUnnormalizedNForQuaternion(const Vec4P& q) {
    const Vec4P e = q/2;
    const RealP ne1 = -e[1], ne2 = -e[2], ne3 = -e[3];
    return Mat43P( ne1,  ne2,  ne3,
                    e[0], e[3], ne2,
                    ne3,  e[0], e[1],
                    e[2], ne1,  e[0]);
}
``` 
(found lurking in Simbody). Note that the normalization status of the quaternion must be carefully considered; usually they are not normalized when this is needed so a scaled qdot is required for correctness.

sherm1, 2016-11-18:
@mitiguy volunteered to fix this. Paul, please take a look at the inverse method also since it should be similarly trivial and lossless but may be similarly baroque.

mitiguy, 2016-11-22:
I'll enjoy working on this.

Paul

edrumwri, 2016-11-22:
Ah, didn't realize that Paul already volunteered to fix it. Paul, I've got a working fix for this already that loses 2.2 digits of precision instead of 5. It needs a little more documentation but is 90% complete. Do you think you can get more precision than that?

sherm1, 2016-11-23:
One issue to consider carefully is the expressed-in frame for both the angular velocity and the vector portion of the quaternion. A quaternion gives the orientation of frame B in A, equivalent to the rotation matrix R_AB, but stored (roughly speaking) in an (angle,axis) format, likely with the axis expressed in A. Given an angular velocity vector w_AB representing the angular velocity of B in A, the conversion to quaternion derivative will expect w_AB to expressed in A if the rotation axis is in A. For a joint where A is the parent body and B the child, that means w_AB would be expressed in the parent frame. Different conventions could conceivably be in effect! Tread cautiously!

tkoolen, 2016-11-23:
@sherm1, good call, that is indeed a mistake in the PR.

sherm1, 2016-11-23:
After discussion with @mitiguy, here is a clarification:

When converting rotation R_AB to angle/axis or quaternion, by construction the resulting axis vector has the same representation when expressed in *either* frame A or B! So the conversion to quaternion is unambiguous. However, the conversion from angular velocity w_AB to quaternion derivative does depend on the expressed-in frame for w.

sherm1, 2016-11-23:
While I have it handy, here are the angular velocity-to-quaternion derivative conversion matrices for angular velocity in either frame. Say we have a rotation matrix R_AB represented as a quaternion `q=[q0 q1 q2 q3]` where q0 is the scalar element. We are given the angular velocity of B in A as a vector w_AB, expressed in A (w_AB_A) or expressed in B (w_AB_B). Then:
```
      [-q1 -q2 -q3]
qdot= [ q0  q3 -q2] * w_AB_A / 2    (compare with code snippet above)
      [-q3  q0  q1]
      [ q2 -q1  q0]
```
and
```
      [-q1 -q2 -q3]
qdot= [ q0 -q3  q2] * w_AB_B / 2    (contrast with code snippet above)
      [ q3  q0 -q1]
      [-q2  q1  q0]
```

mitiguy, 2016-11-24:
I think we should get very high precision (near 1E-15) if the tests is::
a.  Calculate qdot from q, w.
b.  Calculate w from q, qdot.
c.  Compare  calculated w with initial w converting from w to qdot
Perhaps I can review with Evan to see if I can understand code in-situ.

amcastro-tri, 2016-11-28:
@sherm1, the equations above for the angular velocity expressed in different frames seem to be the same? if `q` represents `R_AB` then the first version and the second one is missing a multiply on the right o f the 4x3 matrix by `R_AB`?

sherm1, 2016-11-28:
> the equations above for the angular velocity expressed in different frames seem to be the same? 

No, the off-diagonal signs are flipped in the lower 3x3 part of the matrix. That's all it takes!

edrumwri, 2017-01-31:
Addressed by PR #4965.

closed 2017-01-31 by edrumwri