RobotLocomotion/drake #4126: Contact model numerical issues

SeanCurtis-TRI, 2016-11-11:
The contact model in `RigidBodyPlant` has some issues:  

   1) It uses an arbitrary value of EPSILON which doesn't follow style guide and is very _ad hoc_.
   2) This same code computes a frame aligned with a normal.  The actual code seems suspect and may not be numerically robust.  This should be revisited.

sherm1, 2016-11-11:
In case a better frame-making method is needed, here is one that I believe to be numerically robust. This one assumes that `this` is already a unit vector u, and returns a perpendicular unit vector. One more cross product yields a complete frame. 
```c++
template <class P, int S> inline UnitVec<P,1> 
UnitVec<P,S>::perp() const {
    // Choose the coordinate axis which makes the largest angle
    // with this vector, that is, has the "least u" along it.
    const UnitVec<P,1> u(abs());    // reflect to first octant
    const int minAxis = u[0] <= u[1] ? (u[0] <= u[2] ? 0 : 2)
                                     : (u[1] <= u[2] ? 1 : 2);
    // Cross returns a Vec3 result which is then normalized.
    return UnitVec<P,1>( *this % UnitVec<P,1>(minAxis) ); // % is cross product
}
```
(Code is from Simbody.)

SeanCurtis-TRI, 2016-11-17:
Because I'm too lazy to look myself; `UnitVec<P,1>(int i)` produces a unit vector along either the x, y, or z axes based on i = 0, 1, or 2, respectively?

sherm1, 2016-11-17:
Right!

SeanCurtis-TRI, 2016-11-30:
#4272 is merged and solves this problem.

closed 2016-11-30 by SeanCurtis-TRI