RobotLocomotion/drake #4131: Clean up Drake linking to support nightly builds

david-german-tri, 2016-11-14:
From discussion on Slack today: All fine-grained Drake-internal libraries should be static.  These static libraries should be used as building blocks for a small number of dynamic libraries that are release products.  We can get a good start with just one such dynamic library, `libDrake`, and add smaller libraries as needed.

This issue contributes to #1766

david-german-tri, 2016-11-14:
From further discussion on drake: Examples and regression tests should link to libDrake.  Unit tests, and utility executables that don't represent code a user might write, should just link to the particular static libraries they need.

EricCousineau-TRI, 2018-02-08:
@jwnimmer-tri Is this issue closeable (is there a better / more up-to-date issue elsewhere), or should we keep this on the table for maintaining private linking?

jwnimmer-tri, 2018-02-08:
Yeah, I'm not sure this issue has any value anymore.

closed 2018-02-08 by jwnimmer-tri