RobotLocomotion/drake #4133: Adding functionality to Drake

aespielberg, 2016-11-14:
Hi all,

We have been developing some (what I think is) useful code for soft body simulation that I'd like to push back to drake.  Right now it works almost completely separately, but we are developing an interface to let it interface easily with rigid body dynamics.

What is the best way to go about integrating this into Drake?  Are there any guides for integration and testing, other than the formal code review process?

liangfok, 2016-11-14:
Thanks for contributing back to Drake! Yes, being familiar with the [code review process]() is a must. Other than that, one idea is to incrementally to do the integration by defining a sequence of spirals. See some examples below:

* https://github.com/RobotLocomotion/drake/issues/3994
* https://github.com/RobotLocomotion/drake/issues/3919
* https://github.com/RobotLocomotion/drake/issues/3910

By defining the spirals, we can provide quick architectural-level feedback. The resulting incremental PRs will also be easier and thus faster to review.

david-german-tri, 2016-11-14:
As @liangfok said, please open an issue that describes the design, and explains the spiral of PRs necessary to implement it.  You'll probably get some important feedback from @amcastro-tri, among others.

Once that's done and you're ready to start creating PRs, check out http://drake.mit.edu/developers.html#review-process.  Regarding testing in particular, reviewers will ask that your new code has fine-grained, quick-to-execute unit tests that protect all the important new behaviors.

aespielberg, 2016-11-15:
Thank you guys for the advice.  This is really good to know for how to get
something tested and integrated long-term in Drake.

Playing around today, though, I think I can hone my question a little
better (and perhaps I should make a new thread?).  I have an external that
I want to add to Drake, which I then want to wrap inside Drake to be used
there.  However, the main problem is adding the external itself.  I've been
playing around with the CMakeLists.txt, but I still seem to be getting an
error as I do this.  Is there a guide for all the steps that should be
taken in order to add an external?

-Andy S.

liangfok, 2016-11-15:
To my knowledge, there's currently no written documentation on how to add an external. It'll probably be easiest if your external is a git repository because then it can be added as a submodule in `drake-distro/externals`. Then use [drake_add_external](https://github.com/RobotLocomotion/drake/blob/master/cmake/externals.cmake#L364) to include it in the superbuild.

aespielberg, 2016-11-15:
Hmm, this is exactly what I had tried.  I have a solution which builds
completely fine in isolation, but when I try to include it, I'm getting
errors.  I should note that I am trying to build on Windows (checkout from
before last windows SHA), and I know this is unsupported so I shouldn't
expect a ton of help here, but I don't think this is necessarily the source
of the error.

Is there an issue with trying to add an external solution which itself has
subprojects?  Do I essentially need to unfold those subprojects?

-Andy S.

david-german-tri, 2016-11-15:
> Is there an issue with trying to add an external solution which itself has
subprojects?  Do I essentially need to unfold those subprojects?

You shouldn't need to do that, no.  The Drake superbuild initializes submodules recursively.  

> I'm getting errors.  I should note that I am trying to build on Windows (checkout from
before last windows SHA), and I know this is unsupported so I shouldn't
expect a ton of help here...

I personally am not up for unwinding a CMake failure from an old SHA on Windows, but if you post your error messages you might possibly get input from someone.  That said, your changes will have to work at HEAD of master on Ubuntu before we can merge them, so you might want to consider biting the bullet and developing there locally.

liangfok, 2016-11-15:
>  I have a solution which builds completely fine in isolation, but when I try to include it, I'm getting errors.

One strategy for solving this problem is to build in verbose mode and compare the build commands when building your solution in isolation versus as a Drake external.

Using make:

```
make VERBOSE=true
```

Using ninja:

```
ninja -v
```

aespielberg, 2016-11-15:
Yes, I will absolutely make sure this is up to date and compiling with Ubuntu come the time I push it back.  For now, for various reasons, it's easiest I work on the Windows build.

For a minimal-ish example of what's wrong, here's my current situation.

I have an external called "robot" that I wish to add (poorly named).  The file structure looks like this:

```
  -robot
   CMakeLists.txt
    -projects
      -external_voxel
       Various code in here
       CMakeLists.txt
    -externals
     Various code in here
```

So the main CMakeLists.txt in here creates a project and includes references to the externals and the project external_voxel under projects.

The CMakeLists, for robot:
```
cmake_minimum_required(VERSION 3.0)

project (robot)

# Check Visual Studio version number.
if (MSVC12)
	set(MSVC_LIB_EXTENSION vs2013)
elseif (MSVC14)
	set(MSVC_LIB_EXTENSION vs2015)
endif (MSVC12)

find_library(GTEST_LIB_DEBUG
	"gtestd_${MSVC_LIB_EXTENSION}"
	PATHS ${PROJECT_SOURCE_DIR}/externals/googletest/lib/Debug
	NO_DEFAULT_PATH
)
find_library(GTEST_LIB_RELEASE
	"gtest_${MSVC_LIB_EXTENSION}"
	PATHS ${PROJECT_SOURCE_DIR}/externals/googletest/lib/Release
	NO_DEFAULT_PATH
)
find_library(GTEST_MAIN_LIB_DEBUG
	"gtest_maind_${MSVC_LIB_EXTENSION}"
	PATHS ${PROJECT_SOURCE_DIR}/externals/googletest/lib/Debug
	NO_DEFAULT_PATH
)
find_library(GTEST_MAIN_LIB_RELEASE
	"gtest_main_${MSVC_LIB_EXTENSION}"
	PATHS ${PROJECT_SOURCE_DIR}/externals/googletest/lib/Release
	NO_DEFAULT_PATH
)
find_library(FREEGLUT_LIB_DEBUG
	freeglutd
	PATHS ${PROJECT_SOURCE_DIR}/externals/freeglut/lib/x64
	NO_DEFAULT_PATH
)
find_library(FREEGLUT_LIB_RELEASE
	freeglut
	PATHS ${PROJECT_SOURCE_DIR}/externals/freeglut/lib/x64
	NO_DEFAULT_PATH
)
find_library(ANTTWEAKBAR_LIB
	AntTweakBar64
	PATHS ${PROJECT_SOURCE_DIR}/externals/AntTweakBar/lib
	NO_DEFAULT_PATH
)

include(${PROJECT_SOURCE_DIR}/projects/external_voxel/CMakeLists.txt)
```

and for the one under external_voxel:
```
set(VOXEL_SOURCE_DIR ${PROJECT_SOURCE_DIR}/projects/external_voxel)

file(GLOB_RECURSE VOXEL_HEADER ${VOXEL_SOURCE_DIR}/*.h)
file(GLOB_RECURSE VOXEL_SOURCE ${VOXEL_SOURCE_DIR}/*.cpp)

add_library(external_voxel STATIC ${VOXEL_HEADER} ${VOXEL_SOURCE})
target_include_directories(external_voxel PRIVATE ${VOXEL_SOURCE_DIR}/include)
target_include_directories(external_voxel PRIVATE ${VOXEL_SOURCE_DIR}/include/vecmath)
target_compile_definitions(external_voxel PRIVATE _CRT_SECURE_NO_WARNINGS)

set_target_properties(external_voxel
	PROPERTIES
	OUTPUT_NAME voxel
	DEBUG_POSTFIX "d_${MSVC_LIB_EXTENSION}"
	RELEASE_POSTFIX "_${MSVC_LIB_EXTENSION}"
)
```

When I try to build the superbuild, I get the following error:

```
Severity	Code	Description	Project	File	Line	Suppression State
Error	MSB1009	Project file does not exist.	robot	C:\ResearchCode\drake-soft\artifacts\MSBUILD	1	
```

or for more verbose output:

```
 Microsoft (R) Build Engine version 14.0.25420.1
11>  Copyright (C) Microsoft Corporation. All rights reserved.
11>
11>MSBUILD : error MSB1009: Project file does not exist.
11>  Switch: install.vcxproj
```

I know this is the VS compiler I'm using, but since this project compiles fine in isolation and only chokes when I try to merge it with drake, I suspect that is not the problem.  Googling has given me sparse help.

Does anyone see something obvious I'm doing wrong?

aespielberg, 2016-11-15:
For reference, this is how I edited the main CMakeLists.txt to be:

```
cmake_minimum_required(VERSION 3.5)
project(drake-superbuild)

include(CTest)

include(cmake/config.cmake)
include(cmake/externals.cmake)
include(cmake/examples.cmake)
include(cmake/git/hooks.cmake)

drake_setup_superbuild()
drake_setup_platform()
drake_setup_git_hooks()

###############################################################################
# BEGIN options

##########################################
# External Projects that are ON by default
##########################################
option(WITH_EIGEN "required c++ matrix library.  only disable if you have it already." ON)
option(WITH_GOOGLETEST "required c++ unit test library.  only disable if you have it already." ON)
option(WITH_GFLAGS "required c++ command-line library.  only disable if you have it already." ON)
option(WITH_GOOGLE_STYLEGUIDE "provides cpplint.py style checking" ON)
option(WITH_SWIGMAKE "helper tools to build python & MATLAB wrappers for C++ libraries with Eigen" ON)
option(WITH_BULLET "used for collision detection" ON)
option(WITH_LCM "interprocess communications protocol for visualizers, etc" ON)
option(WITH_BOT_CORE_LCMTYPES "required LCM types library. only disable if you have it already." ON)
option(WITH_SPDLOG "spdlog text logging facility; disabling will turn off text logging." ON)
option(WITH_ROBOT "FEM Library for soft robotics." ON)
if(WIN32)
  option(WITH_GTK "precompiled gtk binaries/headers for Windows" ON)  # needed for lcm on windows
else()
  option(WITH_DIRECTOR "vtk-based visualization tool and robot user interface" ON) # not win32 yet.  it builds on windows, but requires manually installation of vtk, etc.  perhaps make a precompiled director pod (a bit like snopt)
  option(WITH_LIBBOT "simple open-gl visualizer + lcmgl for director" ON) # there is hope, but i spent a long time cleaning up c/c++ language errors trying to make msvc happy.. and still had a long way to go.
  option(WITH_NLOPT "nonlinear optimization solver" ON)
  option(WITH_DREAL "nonlinear SMT solver" ON)
  option(WITH_MOSEK "convex optimization solver; free for academics" OFF)
  # IPOPT is currently disabled on Mac due to MATLAB compatibility
  # issues: https://github.com/RobotLocomotion/drake/issues/2578
  if (NOT APPLE)
    option(WITH_IPOPT "nonlinear optimization solver" ON)
  endif()
  option(WITH_SWIG_MATLAB "A version of SWIG with MATLAB support" ON)
endif()

option(WITH_SNOPT_PRECOMPILED "precompiled binaries only for snopt; the source requires a license (will be disabled if WITH_SNOPT=ON)" ON)
option(WITH_YAML_CPP "library for reading and writing yaml configuration files" ON)

##############################################################
# External Projects that are only needed when MATLAB is in use
##############################################################
include(CMakeDependentOption)
option(DISABLE_MATLAB "Don't use MATLAB even if it is present." OFF)
if(DISABLE_MATLAB)
  message(STATUS "MATLAB is disabled.")
else()
  find_program(matlab matlab)
  if(matlab)
    message(STATUS "Found MATLAB at " ${matlab})
  else()
    message(STATUS "Looked for MATLAB but could not find it.")
  endif()
endif()

## The following projects are default ON when MATLAB is present and enabled.
## Otherwise, they are hidden and default OFF.
[... 300 lines cut ...]

# END examples
###############################################################################

## grab and install precompiled snopt

# TODO: look for snopt_c
if(snopt_c_FOUND OR WITH_SNOPT)
  set(WITH_SNOPT_PRECOMPILED OFF)
endif()
if(WITH_SNOPT_PRECOMPILED)
  message(STATUS "Preparing to install precompiled snopt")
  ExternalProject_Add(download-snopt-precompiled
    URL "https://s3.amazonaws.com/drake-provisioning/drakeSnopt.zip"
    URL_MD5 "7b36168cba2fb9a56b2fd6117427fc4a"
    SOURCE_DIR "${CMAKE_BINARY_DIR}/snopt-precompiled"
    CONFIGURE_COMMAND ""
    BUILD_COMMAND cmake -E copy_directory
      ${CMAKE_BINARY_DIR}/snopt-precompiled/
      ${PROJECT_SOURCE_DIR}/drake/matlab/solvers/
    INSTALL_COMMAND "")
  add_dependencies(download-all download-snopt-precompiled)
  add_dependencies(drake download-snopt-precompiled) # just in case: make sure any compiled drake version happens after precompiled install
endif()
```

amcastro-tri, 2016-11-17:
Hello @aespielberg, glad to hear you are working on this!. I am actually doing some research myself on how to do soft body simulation within Drake. I am curious to know what approach you used, if applicable to general 3D configurations and about its integration with rigid body dynamics. 
As the guys told you above probably a good way to start would be opening and issue where we could spiral a discussion and then a series of smaller PR's.
Thank you!

RussTedrake, 2016-11-18:
Thanks @aespielberg .  but, uh.  Can we not call the new external WITH_ROBOT?  :)  What external package is it exactly?

aespielberg, 2016-11-18:
Hahaha, yes, WITH_ROBOT is a terrible, terrible name.  We're going to fix it once we fix the cmake stuff.  I will provide more details about it on slack tomorrow for you and @amcastro-tri .

RussTedrake, 2016-11-27:
This discussion simmered.  Will close this for now.

closed 2016-11-27 by RussTedrake