RobotLocomotion/drake #4137: Expected control rate for atlas examples

samkhal, 2016-11-16:
When I run `runAtlasWalking` (from the last matlab-supported SHA) and time the plan and control steps, I get rates on the order of 20 Hz (0.01s for plan, 0.04s for control). I haven't started benchmarking everything yet, but in the relevant papers there were rates of 333 Hz or 1 kHz mentioned (though I don't recall the computer specs).

I'm running on a 2.8 GHz i7-4700MQ on a laptop, with one core maxing out to run this. Are these rates expected? Is there anything I should look at in terms of boosting performance?

RussTedrake, 2016-11-18:
on the robot, we run each component as a separate executable (communicating over LCM), and of course on a fairly beefy desktop.

closed 2016-11-18 by RussTedrake

samkhal, 2016-11-28:
Can you share the specs of the computer it was running on? I don't imagine separating the process would affect this question since I'm benchmarking the control process individually. I'm just curious as to why I'm running an order of magnitude slower than you guys were.

hongkai-dai, 2016-11-28:
Are you compiling in debug mode?

samkhal, 2016-11-28:
No, my build type is Release.

hongkai-dai, 2016-11-29:
When you time the control cycle, do you include the simulation time, or only solving the QP?

samkhal, 2016-12-06:
I'm not including simulation; I'm just timing the function `InstantaneousQPController.updateAndOutput`.

The timing code is already there; I'm just unsuppressing output by setting `quiet = false` in `BipedPlanEvalAndControlSystem.m`