RobotLocomotion/drake #4142: SNOPT download failure

aespielberg, 2016-11-16:
On Ubuntu.  What could cause this error?

```Performing download step (download, verify and extract) for 'download-snopt-precompiled'
-- verifying file...
       file='/home/andrews/ResearchCode/drake-soft/artifacts/download-snopt-precompiled-prefix/src/drakeSnopt.zip'
-- MD5 hash of
    /home/andrews/ResearchCode/drake-soft/artifacts/download-snopt-precompiled-prefix/src/drakeSnopt.zip
  does not match expected value
    expected: '7b36168cba2fb9a56b2fd6117427fc4a'
      actual: 'd41d8cd98f00b204e9800998ecf8427e'
-- File already exists but hash mismatch. Removing...
-- Downloading...
   dst='/home/andrews/ResearchCode/drake-soft/artifacts/download-snopt-precompiled-prefix/src/drakeSnopt.zip'
   timeout='none'
-- Using src='https://s3.amazonaws.com/drake-provisioning/drakeSnopt.zip'
-- Retrying...
-- Using src='https://s3.amazonaws.com/drake-provisioning/drakeSnopt.zip'
-- Retry after 5 seconds (attempt #2) ...
-- Using src='https://s3.amazonaws.com/drake-provisioning/drakeSnopt.zip'
-- Retry after 5 seconds (attempt #3) ...
-- Using src='https://s3.amazonaws.com/drake-provisioning/drakeSnopt.zip'
-- Retry after 15 seconds (attempt #4) ...
-- Using src='https://s3.amazonaws.com/drake-provisioning/drakeSnopt.zip'
-- Retry after 60 seconds (attempt #5) ...
-- Using src='https://s3.amazonaws.com/drake-provisioning/drakeSnopt.zip'
CMake Error at download-snopt-precompiled-prefix/src/download-snopt-precompiled-stamp/download-download-snopt-precompiled.cmake:157 (message):
  Each download failed!
```

jamiesnape, 2016-11-16:
SNOPT precompiled is no longer supported.

closed 2016-11-16 by jamiesnape

RussTedrake, 2016-11-18:
i guess that means our peg revisions are unable to use this now, too?

jamiesnape, 2016-11-18:
True. Would you like me to back-port a patch and tag the repo?

RussTedrake, 2016-11-19:
if it's not too difficult to maintain, yes.  the "last windows" sha is probably the most notable reason.