RobotLocomotion/drake #4146: uninitialized warning on gcc 5.4 -m32

BetsyMcPhail, 2016-11-17:
```
In file included from /home/ubuntu/drake/drake/../drake/systems/framework/primitives/gain.h:6:0,
                 from /home/ubuntu/drake/drake/../drake/systems/framework/primitives/gain-inl.h:8,
                 from /home/ubuntu/drake/drake/systems/framework/primitives/test/gain_scalartype_test.cc:1:
/home/ubuntu/drake/drake/../drake/systems/framework/leaf_system.h: In member function ‘void drake::systems::LeafSystem<T>::DoCalcNextUpdateTime(const drake::systems::Context<T>&, drake::systems::UpdateActions<T>*) const [with T = Eigen::AutoDiffScalar<Eigen::Matrix<double, 2, 1, 2> >]’:
/home/ubuntu/drake/drake/../drake/systems/framework/leaf_system.h:264:75: error: ‘*((void*)&<anonymous> +16)’ is used uninitialized in this function [-Werror=uninitialized]
         std::numeric_limits<typename Eigen::NumTraits<T1>::Real>::infinity();
                                                                           ^
/home/ubuntu/drake/drake/../drake/systems/framework/leaf_system.h:264:75: error: ‘*((void*)&<anonymous> +24)’ is used uninitialized in this function [-Werror=uninitialized]
cc1plus: some warnings being treated as errors
```

jwnimmer-tri, 2016-11-18:
What's the recipe to repro?

jwnimmer-tri, 2016-12-12:
@BetsyMcPhail how do we observe or reproduce the error, in order to fix it?

BetsyMcPhail, 2016-12-13:
Thanks for the reminder - I will try to get some more information on this error.

david-german-tri, 2017-04-11:
We were originally interested in standing up a Xenial 32-bit build to replace the Windows 32-bit coverage that we turned down in October 2016.  As of November, I believe this error was the only known blocking issue.

I think it would still be nice to get this coverage, but I'm tagging it as low-priority, since we haven't had any particular demand for 32-bit Drake.

EricCousineau-TRI, 2018-11-02:
For the MATLAB versions that we are testing on for our lightweight `call_matlab` interface, I don't believe 32-bit is an option for compilation anymore. Closing for now.

\cc @jamiesnape

closed 2018-11-02 by EricCousineau-TRI