RobotLocomotion/drake #4147: Add stiction model to RigidBodyPlant

sherm1, 2016-11-17:
Per discussion with @siyuanfeng-tri at the Utah offsite, he has an urgent need for stiction (static friction) for his foot-contact application. The current friction model isn't providing the expected behavior.

Ultimately stiction should be modeled by adding and removing a no-slip constraint. However, we can get a fair approximation now using a continuous "creep" model that limits slip velocity to below an arbitrary tolerance while generating the appropriate constraint force.

Related: #1978 (should replace discontinuous normal force model with a smooth one).

amcastro-tri, 2016-11-17:
I can give this a try once #4033 lands. Probably worth defining now the "materials library"?

amcastro-tri, 2016-11-17:
Excellent. @siyuanfeng-tri will provide us with a cleaner PR from #4004 for us to play with.

siyuanfeng-tri, 2016-11-17:
#4152 is that pr.

sherm1, 2016-11-17:
@amcastro-tri, some notes on a creep stiction model:
For a compliant contact we have a normal force model `Fn=Fn(x,xdot)` and a tangential force model `Ft=mu(v)*Fn`. Here Fn and Ft are scalars, with Fn >= 0; x is penetration depth, v is slip speed. The friction model is mu(v), and we want it to look something like this: 
[image]

Ignoring viscous friction for now, the parameters to this model are static and dynamic coefficients of friction mu_s and mu_d, and a "transition velocity" v_t that sets the largest "creep" velocity that is acceptable during stiction (this can be set very small at the cost of increasing stiffness). The simplest model I know of that provides this shape is the Hollars model:
```c++
// CAUTION: uv and v must be dimensionless in multiples of transition velocity.
// Const 9 flops + 1 divide = approx 25 flops.
// This calculates a composite coefficient of friction that you should use
// to scale the normal force to produce the friction force.
// Curve is similar to Stribeck above but more violent with a discontinuous
// derivative at v==1.
inline static Real hollars(Real us, Real ud, Real uv, Real v) {
    const Real mu =   std::min(v, Real(1))*(ud + 2*(us-ud)/(1+v*v))
                    + uv*v;
    return mu;
}
```
(code snatched from Simbody)

That's the basic idea. The Simbody code for friction in a Hertz/Hunt&Crossley contact is [here](https://github.com/simbody/simbody/blob/master/Simbody/src/CompliantContactSubsystem.cpp#L811).

amcastro-tri, 2016-11-17:
Thank you @sherm1, I will study that code and I'll get back to you.

sherm1, 2016-11-17:
P.S. Simbody actually uses a smoother function called `stribeck()`.

amcastro-tri, 2016-11-17:
Thank you. Do you recommend then implementing this `stribeck()` function instead? is it more computationally intensive than `hollars()`?

sherm1, 2016-11-18:
They are both very cheap. stribeck() takes a little more code but runs faster overall (typically 10-20% faster) because its smooth profile lets larger integration step sizes be used during stiction<->sliding transitions. So I do recommend stribeck().

RussTedrake, 2016-11-18:
fwiw -- @mposa used stribeck with some success when examining static friction experimentally on Atlas.

amcastro-tri, 2016-11-22:
I just found out in #4232 that the problem with the Valkyrie simulation is not the contact model (which of course can be improved) but a number of collision elements not being filtered out.

SeanCurtis-TRI, 2016-11-22:
Yep.

amcastro-tri, 2017-02-01:
Delighted to review this @SeanCurtis-TRI. I just un-assigned myself as part of my triage.

PR #5197 (2017-02-22): Hunt crossley contact model
This PR includes the implementation of the new model and the unit tests.  It is the first of a sequence of *three* PRs including:

   - A PR with three examples illustrating the impact of the model and how to use it.
   - A PR with a doxygen module discussing the model, its parameters, and some of the engineering tradeoffs in using it.

Resolves #4147 
Resolves #4305

For reviewers:

It is worth noting that the *meat* of this PR is in `rigid_body_plant.h/cc` (the implementation of the contact model) and `contact_force_formula_test.cc` (the unit tests for said functi [... cut]
closed 2017-02-22 by sherm1