RobotLocomotion/drake #4153: RigidBodyActuator Only Actuates One-DOF

liangfok, 2016-11-17:
[RigidBodyActuator](https://github.com/RobotLocomotion/drake/blob/master/drake/multibody/rigid_body_actuator.h) only actuates upon a single DOF. Is this assumption final? Will we be generalizing it to support multi-DOF actuators (i.e., actuators that span multiple joints)?

amcastro-tri, 2016-11-17:
I think that is correct. Could you please be more specific with an example of application?

liangfok, 2016-11-17:
Consider the four-bar linkage at the bucket-end of the backhoe:

http://www.iowafarmequipment.com/bradco_backhoes.html

Notice how the hydraulic cylinder goes across four joints. How can I model this?

amcastro-tri, 2016-11-17:
Let's see if I understand. What you would do in this case is to model the hydraulic piston as a prismatic joint (still a 1 dof joint). Then a 1dof actuator would actuate that joint by appling a forcing term that, in this case, would be the force in the axial direction (similarly to an actuator on a revolute joint applying a 1dof torque). Does this answer your question?

liangfok, 2016-11-17:
Ah yes, I believe this answers my question. Thanks. I think this issue can be closed once we add an assertion enforcing the requirement that actuators are only constructed for one-DOF joints.

RussTedrake, 2016-11-18:
a differential drive might be a better example where a single actuator might be used for multiple joints.  or perhaps even the schunk parallel gripper that @sammy-tri was working with?

amcastro-tri, 2016-11-18:
Those are great examples @RussTedrake!. I guess what I was trying to say is that right now Drake only seems to support 1DOF actuators (and therefore I think it is probably a good idea that @liangfok adds a check enforcing this for now). Of course we want to support these models you mention. Probably if a need for the schunk parallel gripper, we can push the implementation of a model for that particular actuator. Though if I am not mistaken such a model would require a constraint. A general framework for constraints in RBT is in my TODO list.

closed 2017-02-01 by amcastro-tri