RobotLocomotion/drake #4155: Possible kuka_simulation Performance Regression Between 2016/11/11 and 2016/11/17

liangfok, 2016-11-17:
# Problem Definition
@patmarion reports a possible performance regression of [kuka_simulation.cc]() between the current head of master and version 0eff10cf28f7bc305a03bbd8fac5a449ed311f82. The former is cycling at 200Hz whereas the latter was cycling at around 700-800Hz.

# Action Item

Verify that this is indeed the case; then, if necessary, verify the cause.

jwnimmer-tri, 2016-11-17:
Shot in the dark: the new plant output port in #3821, which if I understand correctly is computed even if not connected.

patmarion, 2016-11-17:
Interesting.  That means that previously collision checking was not enable, but now it is?

Maybe that could also explain a new crash I am seeing in the kuka_sim under certain control inputs:

<pre>
terminate called after throwing an instance of 'std::runtime_error'
  what():  In BulletModel::findClosestPointsBetweenElements: No closest point found between 30920272 and 30933200
Aborted
Core dumped.
</pre>

david-german-tri, 2016-11-17:
> which if I understand correctly is computed even if not connected.

This is true - #2890.

SeanCurtis-TRI, 2016-11-17:
I just did a test that painfully confirms that the computation of the `ContactResults` is *definitely* the culprit.  I'll change the assignee to me and look into it.

patmarion, 2016-11-17:
fyi, the urdf loaded by the kuka sim is the "no collision" version of the urdf, but in fact that urdf *does* have some collision geometry.

SeanCurtis-TRI, 2016-11-17:
So, initial testing has revealed the cause for the slow down (and points towards the issues that need to be addressed.)

tl;dr
1) Values are not being cached.
2) We're using a horrible method for detecting contacts.

Initial details
To perform a single step, `EvalOutput` is getting called >10 times.  Part of this is due to the fact that the `Simulator` uses RK2 integration and part of it is due to the overall structure of the diagram (such that the more that is downstream of the `RigidBodyPlant` the more times it will be invoked.)  There also seems to be a pseudo-random component due to the asynchronous LCM messages.  This will be alleviated when the cache is in place.

It is worth noting, that the horrible collision detection algorithm (O(N^2)) has always been invoked by the `EvalTimeDerivatives` method.  But that is only invoked by the integrator in a well-disciplined way once per context.  However, when we added the `ContactResult` output, that same bad collision detection algorithm was getting blasted multiple times PER `EvalTimeDerivatives` which is how, even though it's the same functionality, it goes from 800 Hz to 200 Hz.

Of these two issues, the caching issue is a slow work in progress (todo: link issue).  This latter problem is actually next on my plate and, while it won't solve the problem of not caching solutions, it should at least reduce the magnitude of this current instance (todo: link issue).

RussTedrake, 2016-11-18:
fwiw -- the crash you are seeing looks like the standard (but misleading) message that we see when a simulation goes unstable (e.g. state contains NaNs).  Probably not bullet's fault.

SeanCurtis-TRI, 2016-11-18:
@RussTedrake What crash?

Another solution to this would be #2890: don't evaluate an output if it's not actually used.  In the long run, all three solutions should be combined.

patmarion, 2016-11-18:
I think @RussTedrake was referencing the crash I posted here: https://github.com/RobotLocomotion/drake/issues/4155#issuecomment-261381542

SeanCurtis-TRI, 2016-11-18:
Oh, yeah.  Thanks @patmarion.  I'd lost sight of that in the painful performance issues.

SeanCurtis-TRI, 2016-11-22:
This is addressed by #4206.  That PR attacks the inefficient contact force computation problem by improving the O(N^2) collision detection leading to a significant performance improvement.  Feel free to take a look.  

The issues of caching and targeted output evaluation are still on the table.

jwnimmer-tri, 2016-12-22:
Over to @sammy-tri for triage / measurements.

sammy-tri, 2016-12-22:
Depending on where I run the sim, I'm seeing 2500-3200hz out of kuka_simulation on current drake master.  That'll do for now.  I don't believe this ticket needs to be kept around to track caching and targeted output evaluation.

closed 2016-12-22 by sammy-tri