RobotLocomotion/drake #4163: Additional unit testing for RK3 integrator

edrumwri, 2016-11-18:
From Reviewable comments for PR #4033, we need to add the following to the RK3 integrator unit test source:

1. A system composed from two independent mass-spring systems using a Diagram. The inclusion of this system will better stress the vector-based weighting approach used in the error controlled integrator. The two outputs from the independent mass-spring dampers can be checked for accuracy using the existing closed form solution check.
  
2. A check that the system from (1) produces the correct dimensional weighting vectors.

sherm1, 2016-11-18:
We also need to try RK3 as the default and see what breaks.

edrumwri, 2016-11-18:
You mean as the default integrator for Simulator?

sherm1, 2016-11-18:
Yes. For now I mean just to test it (it would have segfaulted with the non-conformant multiplies). I would like a variable step integrator to be the actual default but we need to make sure it works for the existing applications first!

sherm1, 2020-02-25:
I believe this is well tested now. Closing.

closed 2020-02-25 by sherm1