RobotLocomotion/drake #4168: Manipulator Equations and dynamics Jacobian in C++

aespielberg, 2016-11-18:
Is there any Manipulator Equation and/or dynamics Jacobian support in C++?

RussTedrake, 2016-11-19:
yes.  you can use autodiff for gradients, and call e.g. getMassMatrix and inverseDynamics to get the components.  look at the manipulator equations mex entry function for an example.

closed 2016-11-19 by RussTedrake

aespielberg, 2016-11-19:
Thanks for the reply.  Would it be possible to give a direct pointer to which file to look at?  There are a lot of mex functions and I'm not sure which one you are referring to.

RussTedrake, 2016-11-20:
https://github.com/RobotLocomotion/drake/blob/master/drake/matlab/systems/plants/%40RigidBodyManipulator/manipulatorDynamics.m#L42

aespielberg, 2016-11-20:
I'm sorry to be a real pain, but this is not making a lot of sense to me.  I'm less interested in how to mex it up into matlab, and more interested in how I can just access the autodiff in C++ directly.

Trying to make a simple example here,

```typedef AutoDiffScalar<VectorXd> AutoDiffDynamicSize;
		typedef DrakeJoint::AutoDiffFixedMaxSize AutoDiffFixedMaxSize;

		auto r_urdf = RigidBodySystem();
		r_urdf.AddModelInstanceFromFile(GetDrakePath() +
			"/examples/Acrobot/Acrobot.urdf", systems::plants::joints::kFixed);

		auto tree = r_urdf.getRigidBodyTree();
		auto q = VectorXd::Zero(tree->number_of_positions()).eval();
		auto v = VectorXd::Zero(tree->number_of_velocities()).eval();
		auto vd = VectorXd::Zero(tree->number_of_velocities()).eval();
		
		auto qa = q.cast<AutoDiffDynamicSize>().eval();
		auto va = v.cast<AutoDiffDynamicSize>().eval();
		auto vda = vd.cast<AutoDiffDynamicSize>().eval();

		KinematicsCache<AutoDiffDynamicSize> kinematics_cache(tree->bodies);
		kinematics_cache.initialize(qa, va);
		tree->doKinematics(kinematics_cache, true);
		auto mass_matrix = tree->massMatrix(kinematics_cache);

		const RigidBodyTree::BodyToWrenchMap<AutoDiffDynamicSize> no_external_wrenches;

		auto inverseDynamics = tree->inverseDynamics<AutoDiffDynamicSize>(kinematics_cache, no_external_wrenches, vda);
		auto jac = autoDiffToGradientMatrix(inverseDynamics, 6);
		
		auto massMatrix = tree->massMatrix<AutoDiffDynamicSize>(kinematics_cache);
		auto biasTerm = tree->dynamicsBiasTerm<AutoDiffDynamicSize>(kinematics_cache, no_external_wrenches);
		

		std::cout << "Mass matrix is " << massMatrix << std::endl;
		std::cout << "Bias term is " << biasTerm << std::endl;
		std::cout << "Jacobian is " << std::endl;
		std::cout << jac << std::endl;
```

Of course this Jacobian making sense would surprise me, because I haven't explicitly autodiff'd the t, or u or anything.  And, as expected, this doesn't work - the jac matrix is always all zeros.

I cannot mentally parse the mex-fu that's going into making to enabling matlab to get the full jacobian.  Ideally, I want to be able to get out H, C, and B matrices that I can just multiply however and get out an expression with gradients intact, but I can't find any examples of this in the code (alternatively, happy with the full gradients of robot.dynamics() ).

Is there a specific example of how to call these functions and get the gradients?