RobotLocomotion/drake #4169: Warning Related to default_model_instance_id

liangfok, 2016-11-18:
# The Warning

```
[170/210] Building CXX object examples/Valkyrie/test/CMakeFiles/valkyrie_ik_test.dir/valkyrie_ik_test.cc.o
In file included from /home/liang/dev/drake-distro-1/drake/examples/Valkyrie/test/valkyrie_ik_test.cc:11:
In file included from /home/liang/dev/drake-distro-1/drake/../drake/multibody/rigid_body_ik.h:9:
/home/liang/dev/drake-distro-1/drake/../drake/multibody/rigid_body_tree.h:289:11: warning: instantiation of variable 'RigidBodyTree<double>::default_model_instance_id_set' required here, but no definition is available [-Wundefined-var-template]
          default_model_instance_id_set) const;
          ^
/home/liang/dev/drake-distro-1/drake/../drake/multibody/rigid_body_tree.h:1164:30: note: forward declaration of template entity is here
  static const std::set<int> default_model_instance_id_set;
                             ^
/home/liang/dev/drake-distro-1/drake/../drake/multibody/rigid_body_tree.h:289:11: note: add an explicit instantiation declaration to suppress this warning if 'RigidBodyTree<double>::default_model_instance_id_set' is explicitly instantiated in another translation unit
          default_model_instance_id_set) const;
          ^
1 warning generated.
```

jwnimmer-tri, 2016-11-21:
This came from @amcastro-tri adding class-templating, and should definitely be fixed.

amcastro-tri, 2016-11-21:
Hopefully these warning will go away once #3987 is fully resolved. If after #3987 these warning are still in place I will take another look.

PR #4706 (2017-01-10): Addresses -Wundefined-var-template warnings.
Resolves #4169. The numerous warnings were getting bothersome to me. So I resolved them in this PR.

---
This change is [[image]](https://reviewable.io/reviews/robotlocomotion/drake/4706)
closed 2017-01-10 by liangfok