RobotLocomotion/drake #4177: Is it possible to use TimeSteppingRigidBodyManipulator with DircolTrajectoryOptimization?

aash29, 2016-11-19:
Or some other kind of trajectory planning?
thanks in advance

hongkai-dai, 2016-11-19:
`DircolTrajectoryOptimization` assumes that the system has a `dynamics` function, as in https://github.com/RobotLocomotion/drake/blob/master/drake/matlab/solvers/trajectoryOptimization/DircolTrajectoryOptimization.m#L84. The `dynamics` function is in the form
```
xdot = plant.dynamics(t, x, u)
```

You cannot use it directly.

For a `TimeSteppingRigidBodyManipulator`, the dynamics depends not only on `t, x, u` (time, state, control), but also on the external force. The dynamics is in the following form, with extra dependence on external force `lambda`
```
M * vdot + c(q,v) = B*u + J' * lambda
```
So you will need to define a different function to add the dynamics constraint. A good reference is in `ContactImplicitTrajectoryOptimization.m`, https://github.com/RobotLocomotion/drake/blob/master/drake/matlab/solvers/trajectoryOptimization/ContactImplicitTrajectoryOptimization.m#L62~#L127

Notice that you may also need to add constraints on the external force (friction cone, limit surface, etc).

There is another planner that also works with `TimeSteppingRigidBodyManipulator`, check out https://github.com/RobotLocomotion/drake/blob/master/drake/matlab/solvers/trajectoryOptimization/SimpleDynamicsFullKinematicsPlanner.m

RussTedrake, 2016-11-20:
fwiw -- we often do the plan on the RigidBodyManipulator, then play it back with the TimeSteppingRigidBodyManipulator.  The time-stepping models are good for simulation and for some forms of control design, but it's not always the appropriate abstraction for planning.

closed 2016-11-20 by RussTedrake

aash29, 2016-11-21:
Thank you for answering! 
I tried to use ContactImplicitTrajectoryOptimization,

> 
> function res=fallingBrickLCP
> 
> options.floating = 'quat';
> options.twoD = true;
> 
> options.terrain = RigidBodyFlatTerrain();
> s = 'FallingBrick2.urdf';
> p = TimeSteppingRigidBodyManipulator(s,.01,options);
> p = p.addRobotFromURDF(s,[],[],options);
> x0 = [0;1;rpy2quat(randn(3,1));2;1;rpy2quat(randn(3,1))];
> x0 = p.resolveConstraints(x0);
> prog = ContactImplicitTrajectoryOptimization(p,11,[0 2]);
> prog.solveTraj(0.1);

And got this error

> No appropriate method, property, or field getNumJointLimitConstraints for class TimeSteppingRigidBodyManipulator.

getNumJointLimitConstraints method is present in Manipulator class, but not in TimeSteppingRigidBodyManipulator.