RobotLocomotion/drake #4184: requests/questions about discrete time systems syntax

RussTedrake, 2016-11-20:
following #4183, there are a bunch of small issues/questions/requests that I have about the discrete time logic in system2.0

- [ ] can we make DifferenceState -> DiscreteState everywhere?  and of course get_difference_state(), etc to get_discrete_state()? while I appreciate that modal state is also discrete in a sense, I think that the current naming is inconsistent with the continuous time versions.  I believe the context should have time, continuous state, discrete state, modal state, and parameters.  
- [ ] EvalDifferenceUpdate and DoEvalDifferenceUpdate is inconsistent with EvalTimeDerivatives workflow.  i believe I should implement the EvalDifferenceUpdate method in my subclass.

why is ContinuousState a single BasicVector and DifferenceState a list of BasicVectors?
this has the following consequences:
- [ ] DifferenceState is missing get_vector() and get_mutable_vector()
- [ ] DifferenceState is missing operator[]. 
- [ ] Context is missing get_difference_state_vector() and get_mutable_difference_state_vector()

also related:
 - [ ] OutputPort GetMutableData / GetMutableVectorData is inconsistent w/ get_mutable_data, etc from the state classes.

sherm1, 2016-11-20:
Most of these oddities are my fault, Russ. As you surmised, I suggested DifferenceState because modal is also discrete but also to emphasize that the update should be done via difference equations and be limited to updating the xd variables. The API enforces that by providing only xd as mutable, with the rest of the Context const. OTOH, the plan for modal updates is to have them done via the general event handling mechanism which is given a mutable Context and is permitted to update any state variables (continuous, difference/discrete, modal/discrete). If you think calling xd "difference" variables is too pedantic I would be OK with switching them back to "discrete" and clarifying with documentation instead.

The reason xd is a vector of vectors (also my fault) is that I felt that a large discrete System is likely to have variables with different update rates, and with different computational dependencies. Allocating them in groups allows those groups to have their own rates, and allows cache entries to register dependencies on subsets of xd rather than on the whole thing.  Do you think that's overkill? The alternative means that *all* xd-dependent computations are invalidated when *any* xd variable changes. OTOH, other than the broad xc=q,v,z groupings the continuous variables change simultaneously so subsetting them doesn't seem useful.

I'll add myself to the assignees for this issue.

RussTedrake, 2016-11-20:
I'm afraid I definitely prefer DiscreteState to DifferenceState.  Otherwise we should change ContinuousState to DifferentiableState or something.  I worry that the syntactical oddities have been building up to a burden on developers who don't spend every day with the code.

Sorry if I'm slow, but I don't understand why xd has to be mutable.  Shouldn't it only be set through the simulator the same way that a continuous state gets updated (as an "output" argument from the update method)?

I see your point about having a vector of vectors for the discrete state.  My only complaint, again, is the lack of consistency with manipulating the continuous state.

sherm1, 2016-11-20:
> Sorry if I'm slow, but I don't understand why xd has to be mutable.

Sorry, it's not the xd state that's mutable -- the mutable output of EvalDifferenceUpdates() is the update vector. The actual update is performed in the simulator, analogously to the xc update. But for general event handling it is open season on Context changes.

I think we can make xc/xd consistent enough in the case where there is only a single xd grouping, while still retaining the ability to have multiple xd subgroups.

I'm fine with renaming Difference->Discrete with some clarifying comments.

RussTedrake, 2016-11-20:
all great.  thanks sherm.

david-german-tri, 2017-04-18:
Items (1) and (2) have previously been resolved.  I've separated item (6) into #5878, and will be PRing a fix for items (3), (4), and (5) shortly.

PR #5879 (2017-04-19): Clean up the DiscreteValues accessors.
Remove obsolete references to "state", and add sugar for DiscreteValues with just one group.

@sherm1 for feature review.  cc @RussTedrake 

Fixes #4184.

---
This change is [[image]](https://reviewable.io/reviews/robotlocomotion/drake/5879)
commit 64d857a5 (2017-04-19): Clean up the DiscreteValues accessors.

Remove obsolete references to "state", and add sugar for DiscreteValues with just one group.

Fixes #4184.
closed 2017-04-19 by david-german-tri