RobotLocomotion/drake #4187: need DoToAutoDiff for RigidBodyPlant

RussTedrake, 2016-11-20:
Given the new paradigm, we can't actually do control design or planning on multibody systems until this gets implemented.  That's pretty painful.

amcastro-tri, 2016-11-21:
This is very related to the implementation of `RBT::DoToAutoDiff()`, issue #4044 definitely already on my radar. However before getting here this was my proposed line of action:

- Finish templatization of RBT on `<T>`, #3987. This right now requires at least the templatization of joints, followed by 
- making template methods on `<Scalar>`  class methods on `<T>` (getting rid of per-case explicit instantiations) and finally 
- templatization of parameters (essentially change `double` for `<T>` everywhere in RBT and the family).

The idea is that `RBT::DoToAutoDiff()` will essentially be some sort of specialized `clone()` method that "clones" an RBT by calling `RigidBody::DoToAutoDiff()` and `Joint::DoToAutoDiff()`.

Is this proposed line of action sound to you @RussTedrake? how soon do we need this?

RussTedrake, 2016-11-21:
All sounds good.  In some other discussions I raised the possibility of having a second scalartype for parameters (in the context) as being something to consider.  We should decide on that one way or another before you get to bullet 3 above.  +@david-german-tri @sherm1

amcastro-tri, 2016-11-21:
@RussTedrake, what was the context of those discussions? why would we need to have a second scalar type?

RussTedrake, 2016-12-02:
@amcastro-tri asked about autodiff through collision.  Short answer -- getting autodiff for systems that don't have any collision constraints would already be a great start.  We manually specified (approximate) collision gradients before and could do that again.

amcastro-tri, 2017-02-24:
This got duplicated by #4897 and therefore I will close this issue since #4897 is more recent.

closed 2017-02-24 by amcastro-tri