RobotLocomotion/drake #4188: lcm vector gen scripts do not run on mac

RussTedrake, 2016-11-20:
for many reasons, in fact.

First, because the readlink command takes different arguments.
```
drake008% ./pendulum_gen.sh
readlink: illegal option -- f
usage: readlink [-n] [file ...]
./pendulum_gen.sh: line 12: ./tools/lcm_vector_gen.sh: No such file or directory
./pendulum_gen.sh: line 14: gen_vector: command not found
```
the best solution, I think is `greadlink` which I did have installed:
http://stackoverflow.com/questions/1055671/how-can-i-get-the-behavior-of-gnus-readlink-f-on-a-mac

second, they assume clang-format-3.7 (version hard-coded).  homebrew provides 3.8 (and the executable is only named clang-format)

even then, i get some errors:
```
Traceback (most recent call last):
  File "/Users/russt/drake-distro/drake/tools/lcm_vector_gen.py", line 399, in <module>
    main()
  File "/Users/russt/drake-distro/drake/tools/lcm_vector_gen.py", line 396, in main
    generate_code(args)
  File "/Users/russt/drake-distro/drake/tools/lcm_vector_gen.py", line 335, in generate_code
    with open(os.path.join(cxx_dir, "%s.h" % snake), 'w') as hh:
IOError: [Errno 2] No such file or directory: '/Users/russt/drake-distro/drake/examples/Acrobot/gen/acrobot_state_vector.h'
```

n.b. the script also assumes that the following are installed (not covered by our installation instructions, on mac nor linux):
 - clang-format-3.7 
 - pyyaml
should we add them?

RussTedrake, 2016-11-20:
update: the last error above was resolved by manually creating the gen directory.  now everything runs, with everything installed, and local edits to make clang-format-3.7 point to clang-format in lcm_vector_gen.sh and readlink changed to greadlink.

jwnimmer-tri, 2016-11-20:
Re: clang-format... since the install instructions on Ubuntu suggest symlinking `clang-format-##` to be available as `clang-format`, I suspect our scripts should make that the default (there's already an environment variable a user can set to use a different name).

Re: pyyaml -- both 14.04 and 16.04 have `python-yaml`, which I think works?  Maybe only OSX instructions are missing it?

rpoyner-tri, 2016-11-21:
I'll contact Soonho to get an account on the shared Mac, and have a look. Could happen tomorrow or so.

rpoyner-tri, 2016-11-21:
Observations so far:

 * greadlink is part of 'coreutils' under homebrew ; I'll add it to the 'brew install' list.
 * PyYAML is installed via pip already in the homebrew instructions.
 * I can see no good reason for the default spelling of clang-format to have a version number

Having reviewed/patched the above issues, I now see identical yaml parser failures on both mac and ubuntu. I suspect these to be unrelated to this issue, but I think I will PR a fix for yaml errors on ubuntu first just to keep the issues separate.

rpoyner-tri, 2016-11-21:
See prerequisite PR #4196.

closed 2016-11-23 by RussTedrake