RobotLocomotion/drake #4191: System 2.0 API Requires SetZeroConfiguration Boilerplate

liangfok, 2016-11-21:
There's presently a need to set the zero configuration of a plant after the instantiation of a `Simulation` to avoid feeding `NaN` values through the plant's controller. It looks something like the following (the following code was taken from `car_sim_lcm`):

```
  // TODO(liangfok): Modify System 2.0 to not require the following
  // initialization.
  //
  // Zeros the state and initializes controller state.
  systems::Context<double>* controller_context =
      diagram->GetMutableSubsystemContext(
          simulator.get_mutable_context(), controller);
  controller->SetDefaultState(controller_context);

  // TODO(liang.fok): Modify System 2.0 to not require the following
  // initialization.
  //
  // Zeros the rigid body plant's state. This is necessary because it is by
  // default initialized to a vector a NaN values.
  RigidBodyPlant<double>* rigid_body_plant =
      dynamic_cast<RigidBodyPlant<double>*>(controller->plant());
  Context<double>* plant_context =
      controller->GetMutableSubsystemContext(controller_context,
                                             rigid_body_plant);
  rigid_body_plant->SetZeroConfiguration(plant_context);
```

Similar code is being added in #4107.

We should modify System 2.0's API to not require this boilerplate.

jwnimmer-tri, 2016-11-21:
FWIW, I think I was the one who defected this and asked for the TODO.  My hope is that the `System` abstraction will offer a "starting sim now, please populate your t=0 values into this Context param", and that the `Simulator` would invoke that without main-program boilerplate in the common case.

david-german-tri, 2016-11-21:
@jwnimmer-tri Thanks for the clarification.  Seems reasonable to me.

sherm1, 2016-11-21:
We are planning to add an Initialize() call from the Simulator at simulation start.

commit 4f64ccc6 (2016-12-06): address code review.  resolves #4191.  resolves #4352.
closed 2016-12-06 by RussTedrake