RobotLocomotion/drake #4204: Bullet collision detection does not support drake collision filters

SeanCurtis-TRI, 2016-11-21:
Collision FIltering
-----------------------
During collision detection, pairs of bodies are tested to see if they are in collision.  At the limit, there are potentially O(N^2) colliding pairs.  In practice, this is seldom the case.  Collision filtering is a mechanism for excluding pairs from the set of O(N^2) *candidate* pairs such that collision between the pair isn't even considered.  Drake offers two mechanisms for introducing collision filters:

   - Collision cliques: Each collision element can belong two zero or more collision *cliques*.  If two collision elements belong in the same clique, they will not be tested for collision.  This is very useful for doing targeted collision pair filtering (e.g., collision elements which belong to the same rigid body or collision elements which belong to adjacent bodies in a chain.)
   - Filter groups: each collision element can belong to zero or more filter groups.  Furthermore, it can specify zero or more groups that it ignores.  Thus, for collision elements *a* and *b*, the pair, (*a*, *b*) will only be considered for collision if *a* is not a member of *any* group that *b* ignores and vice versa.  This is useful for eliminating large, easily classifiable sets of collision elements (e.g., all the elements of one robot against all the elements of another.)

The two mechanisms can be used in conjunction.

The Problem
-----------------
Bullet offers mechanisms for collision filtering as well.  Generally, it also has filter groups with a very similar semantic and an arbitrary broad-phase callback.  The primary difference between drake's and bullet's filter groups is that bullet is limited to 32 groups, whereas drake supports 128.

The callback allows for arbitrary logic to determine if a pair should be considered for collision.  This appears to be the *ideal* vector for capturing Drake's definition of collision filtering.  However, Drake's usage and Bullets intent do not align. To all appearances, Bullet builds and maintains a conceptual BVH as elements are added to the "collision world".  Unfortunately, we add these elements to the model *before* all of the drake filter definitions are created (or are even available).  Thus, when the bullet filter callback we provide is invoked by bullet, everything is deemed collidable.  And the callback never plays a role in subsequent collision queries.

It would be helpful if @erwincoumans could confirm this.

Ultimately, this comes from trying to build functional data structures as we parse files.  We can't simply deserialize the files directly into classes because some of the data our classes need can only be computed after the full file has been parsed.

The Plan
------------

The logical thing to do in this case is to defer registration of collision elements with the collision model until all collision filter information is known (akin to #2661.)  

Currently, RigidBodyTree performs unnecessary work in building its own collision pairs to account for this Drake-specific logic.  But when we do so, we lose out on bullet's broadphase culling and pay a significant run-time cost.  For example, running `kuka_simulation` with our current collision detection for contact forces produces `IIWA_STATUS` lcm messages at about 220 Hz.  By rolling it over to using Bullet's broadphase culling, we get messages at > 3000 Hz.

The actual solution can be deferred with a simpler approximation.  I propose we do two different PRs.  One that gets the basic performance up (although it's a bit hacky) and a second that more robustly makes use of bullet and drake working together.

1) Modify `RigidBodyTree::ComputeMaximumDepthCollisionPoints`.  At its core, it uses Bullet's `performDiscreteCollisionDetection` and exploits the full broadphase culling.  However, for the reasons above, it will include collision data for pairs that should be filtered.  We can filter the output according to the Drake collision filters and still get most of the performance benefit outlined above.

2) Modify `RigidBodyTree` instantiation to defer registering Collision Element objects with the collision model until *after* the tree has been build and collision cliques and filters have been computed.  This would be done in the `RigidBodyTree::compile` method.  This may also be deferred to when the GeometryWorld is created.

sherm1, 2016-11-21:
Ongoing nag: this is further evidence that we should parse to an intermediate format, then build data structures once we have the full model available.

amcastro-tri, 2016-11-23:
> Thus, when the bullet filter callback we provide is invoked by bullet, everything is deemed collidable. And the callback never plays a role in subsequent collision queries.

Since relevant I will post here the [comment made here](https://reviewable.io/reviews/robotlocomotion/drake/4206#-KXCdJhfZ7vvFF5hEySm).

> I just tested this with Valkyrie and the logic is reached when performDiscreteCollisionDetection when using ComputeMaximumDepthCollisionPoints (reportig about 50 contact pairs). 
Interestingly, Bullet also calls the broadphase callback when using AllPairsClosestPoints (I guess triggred by BulletModel::updateModel()) reporting this time 2853 (N^2) contact pairs.
Another interesting point is that Bullet is not calling the braoadphse callback with each call to performDiscreteCollisionDetection but actually fewer times. This would seem to indicate that Bullet needs to redo its internal broadphase acceleration strucutres once in a while only when it detects that objects are crossing the BVH boundaries.
We can also have a f2f about this if useful and I can also give you the branch I am testing with.

SeanCurtis-TRI, 2016-11-23:
To summarize my response:

We are both right.  My comments are predicated on our initial conditions being wrong.  You are pointing out that over the course of simulation, circumstances can arise that *may* correct the initial conditions.  However, there are no guarantees that the initial conditions will be completely corrected.  So, at the end of the day, we are responsible to make sure the time steps are correct (and then if things break, we can blame Bullet.)

SeanCurtis-TRI, 2016-11-23:
See issue [871](https://github.com/bulletphysics/bullet3/issues/871) in the bullet repo for conversation on the underlying issue.

SeanCurtis-TRI, 2017-01-18:
#4773 finishes out the collision filter logic.

closed 2017-01-18 by SeanCurtis-TRI