RobotLocomotion/drake #4216: parameters should not be protected

RussTedrake, 2016-11-22:
I believe that external methods should be allow to set/get the parameters of a system.  For example, I'm writing a test code for an encoder block which tries to set the calibration offsets (from outside the class).  I would expect something similar to happen with system identification using parameters.

potentially related: it feels inconsistent to access state, etc, via e.g. 
`context->get_mutable_continuous_state()`
 but to have to get to the parameters via 
`this->GetNumericParameters(*context,0)`.

RussTedrake, 2016-11-22:
edit: i found how to set the parameters on a LeafContext<T>.  perhaps this just means that parameters have not been implemented up to the full System<T> level yet?

david-german-tri, 2016-11-22:
Discussed f2f: I'd made a design decision not to expose Parameters on Diagrams, because I couldn't  come up with a reason that a "bag of anonymous parameters" would be useful.  As @RussTedrake pointed out above, it's useful for system identification, so I agree we should do it.  Not too difficult.

david-german-tri, 2017-04-03:
#5467 fixed this.

closed 2017-04-03 by david-german-tri