RobotLocomotion/drake #4220: AffineSystem component matrices should be parameters

naveenoid, 2016-11-22:
Would be useful towards linear sysID. Any opinions? I can work on the fix.

david-german-tri, 2016-11-22:
Seems reasonable to me, and all the framework support you need should already exist.  What sort of opinions are you looking for?

david-german-tri, 2016-11-22:
After a moment's reflection, here's one opinion: I'd suggest keeping the constructor signature as-is, and using the inputs as default parameter values in `AllocateParameters`.

david-german-tri, 2016-11-22:
Also a question: does this suggest we want a subclass of `BasicVector` with matrix semantics?

david-german-tri, 2016-11-22:
(I presume there are Eigen ways to interpret a VectorXd as a MatrixXd with no copies or anything, right?)

RussTedrake, 2016-11-22:
while i agree, maybe i would wait for a use case instead of inventing one?

RussTedrake, 2016-11-22:
i would imagine a potentially better approach to linear system ID is to suck in the data, then write out a new Affine/Linear system.  The order (matrix sizes) is typically an option that the system ID methods would accept.

The case where this would help is if we want to ID some elements of an affine system as a part of a larger diagram.  

But it would also change a lot of the signatures of the existing methods (perhaps another reason to do it sooner).  For instance, the Linearize methods might have to return a context in addition to the actual system?

naveenoid, 2016-11-23:
ok I will close this for now. I agree a good use-case doesnt exist yet.

naveenoid, 2016-11-27:
Getting relevant again in the light of the discussion on Filter/Estimator design

jwnimmer-tri, 2019-03-04:
Given that the reason for re-opening this was "getting relevant again" (and not "Drake is broken"), and that "again" was several years ago, I think it's fair to close this.  Please re-open with a more detailed request, if actually needed.

closed 2019-03-04 by jwnimmer-tri