RobotLocomotion/drake #4224: Include example binaries in the Drake install

david-german-tri, 2016-11-22:
This matters, for instance, to KUKA users.

cc @patmarion

jamiesnape, 2016-11-22:
Also useful for #1766.

patmarion, 2017-03-24:
I would still like to see this happen in the CMake build.  But if there's no plans to do so, please close this issue.

jwnimmer-tri, 2017-03-24:
I think the issue is still valid even some CMake listfiles are going away -- the Bazel binary release could / should include those artifacts.

david-german-tri, 2017-03-24:
I am going to bump the priority down to medium, though.

EricCousineau-TRI, 2018-02-08:
Closing, given that Francois's PRs for a subset of examples have landed.

jwnimmer-tri, 2018-02-08:
I thought the goal was to install ~all examples?

jamiesnape, 2018-02-08:
It is.

jwnimmer-tri, 2020-09-16:
We don't seem to be suffering for not having this since the past four years, and adding more C++ binaries would only increase the image footprint when most users just want the library, not the demos.  Using online notebooks for the demos seems to be the way of the future.

Unless anyone speaks up about this in the next month or two, we'll plan to close this as "won't fix".

closed 2020-12-09 by jwnimmer-tri