RobotLocomotion/drake #4231: Update Gravity Compensation Unit Test To Not Repeat Code Logic

liangfok, 2016-11-22:
From @naveenoid's comment in #4099:

> As an alternative to repeating the computation inside gravity compensation, I'm wondering if it's better to compute the gravity torque, feed it into the system being controlled and then demonstrate that the resulting motion is of 0 acceleration and 0 velocity instead. This might overcome hidden problems in repeating the controller logic as a verification.

Another idea is to simply manually derive the expected gravity comp vector and verify that the output of the `GravityCompensatorSystem` matches this vector.

naveenoid, 2016-11-23:
I think there is also a genuine use case for a single pass (single time step) input to output integration  to verify things like the derivative and output function of systems. For instance, for the gravity compensator (to use the original context of my comment), it might make sense to plug in gravity compensators output in a diagram to a system and then simply evaluate if the system has near zero velocities and accelerations at any given position. This is more or less running a simple algebraic sanity check on controllers without having code repetition in unit tests..This can also be done by constructing the diagram, manually setting state, allocating derivatives and then computing the derivatives inside the unit test. This functionality however might sit inside Simulator.h for instance. Opinions?

liangfok, 2016-11-23:
I like the idea of feeding the `GravityCompensator`'s output into a `RigidBodyPlant` and verifying that the `RigidBodyPlant`'s configuration after a single integrator cycle has near zero velocity and acceleration.

Here's one implementation idea: Create a `GravityCompensatedPlant` class, which extends `Diagram`, and has a constructor that takes as input the model to be gravity-compensated. Provide a `void DoGravityComp(const GravityCompensatedPlant& plant)` method that instantiates a `Simulator` using the `plant`, simulates a single cycle, and then verifies the plant's generalized state vector output's velocity values are near zero.

How do I verify that the plant's acceleration is zero?

naveenoid, 2016-11-23:
Compute the state derivatives and check the top half to be near zero. I can take a swing at this if you would like. It will make a good additional unit test.

liangfok, 2016-11-23:
Yes, by all means take a swing at it! :smile: !

naveenoid, 2017-03-02:
This test is now irrelevant in light of #5339

closed 2017-03-02 by naveenoid