RobotLocomotion/drake #4232: Simulation of Valkyrie: Feet spread apart due to unprocessed collision groups.

amcastro-tri, 2016-11-22:
I found out the reason why the feet in the Valkyrie simulation spread apart.
The reason is that the `urdf` model for Valkyrie in [`valkyrie_A_sim_drake_one_neck_dof_wide_ankle_rom.urdf`](https://github.com/RobotLocomotion/drake/blob/master/drake/examples/Valkyrie/urdf/urdf/valkyrie_A_sim_drake_one_neck_dof_wide_ankle_rom.urdf) contains all the detailed collision geometry using the same grids used for visualization. When Drake loads these grids it actually creates a convex hull of the actual geometry which might result in a bulkier version of that geometry. This however is not the direct cause of the problem.

The problem is that the "collision groups" (as defined for Matlab I believe) defined by the URDF elements `<collision_filter_group>` are not parsed in the C++ version. To be more specific, I believe the reason why the collision group [`l_leg`](https://github.com/RobotLocomotion/drake/blob/master/drake/examples/Valkyrie/urdf/urdf/valkyrie_A_sim_drake_one_neck_dof_wide_ankle_rom.urdf#L3093) is in there, is to avoid the collision between the hip rotational link components and the leg itself. 

An easy test that proves this hypothesis is this: comment out the `<collision>` elements for "leftHipRollLink" and "rightHipRollLink". What happens is that those elements "collide" with the elements forming the actual legs ("leftHipPitchLink" and "rightHipPitchLink" respectively) and then effectively push the legs inwards as observed in the simulation. Removing the `<collision>` tags  for "leftHipRollLink" and "rightHipRollLink" gets rid of the problem. 

A similar issue might be happening with other elements that were intended to be "filtered" by `<collision_filter_group>` tags but their collisions are not that evident in the simulation.

cc'ing @siyuanfeng-tri, @SeanCurtis-TRI, @sherm1

amcastro-tri, 2016-11-22:
How to solve this? it seems that the right thing to do is to parse `<collision_filter_group>` entries as I think is already done in Matlab ( @RussTedrake and @tkoolen might be able to confirm). This parsing then should generate the proper "collision cliques" withing Drake.

siyuanfeng-tri, 2016-11-22:
Yeah, i guessed that today as well. 
I think this is also related to #4204.

SeanCurtis-TRI, 2016-11-22:
It is.  In fact, I'm moved this on my plate next.  It's one of my OKR's and will alleviate a number of collision headaches.

avalenzu, 2016-11-23:
@amcastro-tri, yes those tags are being parsed in MATLAB. The parsing occurs in `addRobotFromURDF.m` ([here](https://github.com/RobotLocomotion/drake/blob/53fd360a6b5c4ae5b225b4d4394845de4bd62dfe/drake/matlab/systems/plants/%40RigidBodyManipulator/addRobotFromURDF.m#L226)) and the information is transferred to the C++ `RigidBodyTree` in `constructModelMex.cpp` ([here](https://github.com/RobotLocomotion/drake/blob/53fd360a6b5c4ae5b225b4d4394845de4bd62dfe/drake/matlab/systems/plants/constructModelmex.cpp#L271))

amcastro-tri, 2016-11-23:
Thank you @avalenzu!. I guess what we need is a similar capability in the C++ implementation. As @SeanCurtis-TRI mentioned, he will be implementing this capability soon. He is now the "all collision related things" guy in our team.

SeanCurtis-TRI, 2016-11-23:
@amcastro-tri Now I want that on a t-shirt.

tkoolen, 2016-11-23:
Are you aware of `RigidBodyTree::removeCollisionGroupsIf`?

SeanCurtis-TRI, 2016-11-23:
I wasn't specifically, but thanks for pointing it out.  I'll definitely factor it in when I'm working on this.

siyuanfeng-tri, 2016-11-30:
just for the laughs, i removed all the collisions in the urdf except the balls on the feet, and the feet still want to move inward. (the sim is a lot faster) i didn't investigate it much further.

amcastro-tri, 2016-11-30:
@siyuanfeng-tri, Could you share the urdf file you are using?

amcastro-tri, 2018-08-14:
This got resolved long ago when we figured out there were a lot of collisions between adjacent bodies. Filtering fixed it as far as I can remember. Old and deprecated.

closed 2018-08-14 by amcastro-tri