RobotLocomotion/drake #4250: Error in the function createFunnelLibrary.m

GitMunF122, 2016-11-24:
Recently I tried to executed the function "createFunnelLibrary.m" in the folder /drake/examples/Quadrotor.
 
Unfortunately, the execution of the function stops with an error (see below), when the program tries to execute line 50 in the file "FunnelLibrary" (which can be found in the folder /drake/systems/trajectories).
 
The complete error message is as follows:
 
Undefined function 'taylorApprox' for input arguments of type 'SimulinkModel'.
 
Error in FunnelLibrary/computeFunnel (line 50)
            sysClPoly = taylorApprox(sysCl, xtraj, [], 3);
 
Error in FunnelLibrary/computeFunnels (line 112)
                obj = obj.computeFunnel(robot, scaleFactor, funnelLength, radius, i);
 
Error in createFunnelLibrary (line 53)
    funLib = funLib.computeFunnels(r, 115, 12, 0.15);
 
The problem here seems to be that the variable 'sysCl' is a 'SimulinkModel', which can not be passed to the function 'taylorApprox'. 'sysCl' is computed two lines above:
sysCl = feedback(robot, tv);
So I don't know if there is something wrong with the function 'taylorApporx', or if the function 'feedback' is being called in a wrong way.
 
Kind regards

RussTedrake, 2016-11-24:
I'm afraid we're not likely to fix this quickly (because it is in matlab), but rather provide a c++ implementation soon.  Will let @psiorx comment if he sees an easy fix.

GitMunF122, 2016-11-28:
I should probably mention that this error also occurs with the current stable release.

jwnimmer-tri, 2017-06-19:
Closing as won't-fix, because #6369 will be removing the relevant software.

If I'm confused and this should in fact stay open, please re-open an clarify the remaining unsolved problem(s) in a new comment here.

closed 2017-06-19 by jwnimmer-tri