RobotLocomotion/drake #4255: Windows Binary Installation constructModelmex.mexw64 specified module could not be found.

andy-yangz, 2016-11-25:
It seems I met similar problem as #1379. I tried its solution, but it doesn't work.
At first, I downloaded the drake win64 version on my win10 computer, I can run some examples well, like the rimless wheel, however, when I run some examples, it will raise some error like.

    >> runPassive
    Invalid MEX-file 'C:\Users\Andy\Documents\MATLAB\drake-v0.9.11-win64\drake\systems\plants\constructModelmex.mexw64': The specified module could not be found.

    Error in RigidBodyManipulator/createMexPointer (line 2139)
        obj.mex_model_ptr = constructModelmex(obj);

    Error in RigidBodyManipulator/compile (line 846)
      model = createMexPointer(model);

    Error in RigidBodyManipulator/addRobotFromURDF (line 60)
      model = compile(model);  % ideally this would happen on entry into any function...

    Error in PlanarRigidBodyManipulator/addRobotFromURDF (line 90)
      model = addRobotFromURDF@RigidBodyManipulator(model,urdf_filename,xyz,rpy,options);

    Error in PlanarRigidBodyManipulator (line 78)
        obj = addRobotFromURDF(obj,urdf_filename,zeros(3,1),zeros(3,1),options);

    Error in runPassive (line 4)
       m = PlanarRigidBodyManipulator('KneedCompassGait.urdf', options);

Before I write this comment, I tried running Dependency Walker inside matlab, and I get this.
[image]
So I search some lost files name on Google, they suggest me do [this](http://stackoverflow.com/questions/17023419/win-7-64-bit-dll-problems).
But I tried all of these, it just doesn't work.
So I wish someone could give me some suggestion about what should I do to solve this problem.

RussTedrake, 2016-11-25:
my guess is that, at least for dependency walker, the library directory which contains DrakeRBM.dll, etc is not in your PATH.

andy-yangz, 2016-11-25:
Solved it. 
It seems like I saved path in matlab before, so some examples can work well.
However, I make few mistakes in my startup.m file, library directory contain DrakeRBM.dll is not in my PATH.
Now it can work well, some animation can be seen, even it goes crazy quickly.
Thank you very much. 

BTW, how to detect problem from dependency walker? it really make me confused.
Now I run dependency walker again, I see the same error message in the Log View.

RussTedrake, 2016-11-25:
you would have to launch dependency walker with the same path as matlab (e.g. i have instructions somewhere on github for launching it directly from the matlab window).

closed 2016-11-25 by RussTedrake