RobotLocomotion/drake #4258: 'QVTKWidget.h' file not found

weiqiao, 2016-11-26:
I'm trying to install drake on macOS 10.12. When I ran "make", I always got error: 'QVTKWidget.h' file not found. The header file 'QVTKWidget.h' should be inside the vtk package if vtk was installed correctly. I installed vtk by first running "brew install cartr/qt4/qt" to install qt4 and then running "brew install vtk5". Is it correct? How to fix the error?

Error message:

[ 81%] Performing build step for 'director'
[  6%] Built target vtkDRCFilters
[ 13%] Built target vtkDRCFiltersPythonD
[ 14%] Built target vtkDRCFiltersPython
[ 14%] Building CXX object src/app/CMakeFiles/ddApp.dir/moc_ddPythonQtDecorators.cxx.o
In file included from /Users/weiqiaohan/Documents/software/drake-distro/build/externals/director/src/director-build/src/app/moc_ddPythonQtDecorators.cxx:9:
/Users/weiqiaohan/Documents/software/drake-distro/build/externals/director/src/director-build/src/app/ddPythonQtDecorators.h:9:10: fatal error: 
      'QVTKWidget.h' file not found
#include "QVTKWidget.h"
         ^
1 error generated.
make[8]: *** [src/app/CMakeFiles/ddApp.dir/moc_ddPythonQtDecorators.cxx.o] Error 1
make[7]: *** [src/app/CMakeFiles/ddApp.dir/all] Error 2
make[6]: *** [all] Error 2
make[5]: *** [src/director-stamp/director-build] Error 2
make[4]: *** [CMakeFiles/director.dir/all] Error 2
make[3]: *** [all] Error 2
make[2]: *** [director-prefix/src/director-stamp/director-build] Error 2
make[1]: *** [CMakeFiles/director.dir/all] Error 2
make: *** [all] Error 2

RussTedrake, 2016-11-26:
per http://drake.mit.edu/homebrew.html , you need to use `brew install vtk5 --with-qt`

weiqiao, 2016-11-26:
When I did "brew install vtk5 --with-qt", I got error even after I reboot the mac:

Error: Operation already in progress for qt
Another active Homebrew process is already using qt.
Please wait for it to finish or terminate it to continue.

RussTedrake, 2016-11-26:
`brew uninstall vtk5` then retry?

weiqiao, 2016-11-26:
I tried uninstalling and reinstalling many times, but it doesn't work.

hongkai-dai, 2016-11-26:
Not sure if it is related, but @patmarion mentioned that the system vtk may be different from the director's version. He recommended me to set the vtk folder by doing
```
$ cd drake-distro/externals/director
$ ccmake .
```
And then change the VTK_DIR option.

weiqiao, 2016-11-26:
I'm able to build drake by turning off the "director" option before "make".

RussTedrake, 2016-11-26:
that’s good to hear, but you’ll want to have the visualizer soon enough.  

i think you should try to resolve the homebrew issue (it’s not director’s fault if you can’t install vtk5 properly on homebrew).  it looks like perhaps you need to zap a lock file, as in: https://github.com/Homebrew/legacy-homebrew/issues/17730 <https://github.com/Homebrew/legacy-homebrew/issues/17730>

weiqiao, 2016-11-26:
Hongkai's suggestion works for me. Thanks!

closed 2016-11-26 by RussTedrake