RobotLocomotion/drake #4259: fortran support just broke on mac?

RussTedrake, 2016-11-26:
symptom:
- running cmake fresh on drake-distro right now fails to find a fortran compiler.  (gfortran is still located in /usr/local/bin/gfortran).  
- calling cmake with `-DCMAKE_Fortran_COMPILER=/usr/local/bin/gfortran` got farther, but now reports 
```
CMake Error at cmake/config.cmake:190 (message):
  GNU Fortran compiler version must be at least 4.9 (detected version )
```
(note the missing detected version)
- disabling fortran with `-DDISABLE_FORTRAN=ON` gets me unstuck, but is obviously not ideal.

cause??
- I *believe* that the only thing that changed is that I installed gdb with homebrew.  

a little more:
- After the errors above started, I did a full brew update, etc, copy and pasting in exactly the hombrew install instructions from sphinx.  Still no change.

I'm thinking perhaps gfortran got updated by homebrew, now doesn't report it's version in the way cmake expects?

RussTedrake, 2016-11-26:
N.B. - it looks like DISABLE_FORTRAN is not passed forward to drake?

jwnimmer-tri, 2016-11-26:
I would have blamed #4244, but it didn't merge yet.  It looks like #4143 changed when Fortran was passed.  Maybe #4244 is required for #4143 to be correct?

RussTedrake, 2016-11-26:
Thanks.  I think #4244 will help with the second issue (not passing DISABLE_FORTRAN to drake), but the fundamental problem is that cmake is not able to find a fortran compiler.  I don’t think either of these issues would contribute to that.

jamiesnape, 2016-11-28:
I certainly need #4244 merged quickly. There was really nothing complicated there that needed such deep review.

jamiesnape, 2016-11-28:
Did you configure after updating the compiler? CMake will look at the old location (it resolves symlinks) and that no longer exists.

jamiesnape, 2016-11-28:
#4143 is also unrelated. Your build would have failed before that regardless.

RussTedrake, 2016-11-28:
this happens even on a clean checkout / build

jamiesnape, 2016-11-28:
Create this `CMakeLists.txt` in a clean directory and run `cmake`:
```cmake
cmake_minimum_required(VERSION 3.5)
project(fortran-test LANGUAGES Fortran)
```
What is the output? You should see something like
```
-- The Fortran compiler identification is GNU 6.2.0
-- Checking whether Fortran compiler has -isysroot
-- Checking whether Fortran compiler has -isysroot - yes
-- Checking whether Fortran compiler supports OSX deployment target flag
-- Checking whether Fortran compiler supports OSX deployment target flag - yes
-- Check for working Fortran compiler: /usr/local/bin/gfortran
-- Check for working Fortran compiler: /usr/local/bin/gfortran  -- works
-- Detecting Fortran compiler ABI info
-- Detecting Fortran compiler ABI info - done
-- Checking whether /usr/local/bin/gfortran supports Fortran 90
-- Checking whether /usr/local/bin/gfortran supports Fortran 90 -- yes
-- Configuring done
-- Generating done
```

RussTedrake, 2016-11-28:
interestingly, cmake generation is also failing now on my work laptop (without any changes to homebrew).  so the fortran error happened in drake, not on that one machine.

Running your test file works, and results in:
```
-- The Fortran compiler identification is GNU 6.1.0
-- Checking whether Fortran compiler has -isysroot
-- Checking whether Fortran compiler has -isysroot - yes
-- Checking whether Fortran compiler supports OSX deployment target flag
-- Checking whether Fortran compiler supports OSX deployment target flag - yes
-- Check for working Fortran compiler: /usr/local/bin/gfortran
-- Check for working Fortran compiler: /usr/local/bin/gfortran  -- works
-- Detecting Fortran compiler ABI info
-- Detecting Fortran compiler ABI info - done
-- Checking whether /usr/local/bin/gfortran supports Fortran 90
-- Checking whether /usr/local/bin/gfortran supports Fortran 90 -- yes
-- Configuring done
-- Generating done
-- Build files have been written to: /Users/russt/tmp/fortran
```

jamiesnape, 2016-11-28:
I am having trouble reproducing it. My setup should be pretty similar to yours. Which version of CMake, Xcode, OS X?

RussTedrake, 2016-11-28:
osx 10.11.6
30-27-187% cmake --version
cmake version 3.6.0

CMake suite maintained and supported by Kitware (kitware.com/cmake).

30-27-187% clang --version
Apple LLVM version 8.0.0 (clang-800.0.42.1)
Target: x86_64-apple-darwin15.6.0
Thread model: posix
InstalledDir: /Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin

jamiesnape, 2016-11-28:
I am the same except for using CMake 3.7 and gfortran 6.2. I will downgrade and see if I can reproduce.

jamiesnape, 2016-11-28:
The error on your work laptop laptop is identical (warning about the Fortran version, with the detected version missing)?

jamiesnape, 2016-11-28:
Also `make` or `ninja`?

RussTedrake, 2016-11-28:
Him.  Maybe it’s actually particular to the Xcode generator.  

```
30-27-187% mkdir build_test
30-27-187% cd build_test
30-27-187% cmake ..
-- The C compiler identification is AppleClang 8.0.0.8000042
-- The CXX compiler identification is AppleClang 8.0.0.8000042
-- Check for working C compiler: /usr/local/opt/ccache/libexec/cc
-- Check for working C compiler: /usr/local/opt/ccache/libexec/cc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Check for working CXX compiler: /usr/local/opt/ccache/libexec/c++
-- Check for working CXX compiler: /usr/local/opt/ccache/libexec/c++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Found Git: /usr/bin/git (found version "2.9.3 (Apple Git-75)")
-- CMAKE_INSTALL_PREFIX=/Users/russt/drake-distro/build_test/install
-- The Fortran compiler identification is GNU 6.1.0
-- Checking whether Fortran compiler has -isysroot
-- Checking whether Fortran compiler has -isysroot - yes
-- Checking whether Fortran compiler supports OSX deployment target flag
-- Checking whether Fortran compiler supports OSX deployment target flag - yes
-- Check for working Fortran compiler: /usr/local/bin/gfortran
-- Check for working Fortran compiler: /usr/local/bin/gfortran  -- works
-- Detecting Fortran compiler ABI info
-- Detecting Fortran compiler ABI info - done
-- Checking whether /usr/local/bin/gfortran supports Fortran 90
-- Checking whether /usr/local/bin/gfortran supports Fortran 90 -- yes
-- Found Matlab: /Applications/MATLAB_R2015a.app/extern/include;/Applications/MATLAB_R2015a.app/simulink/include (found version "8.5") found components:  MAIN_PROGRAM MEX_COMPILER MX_LIBRARY SIMULINK
-- Found Java: /Library/Java/JavaVirtualMachines/jdk1.8.0_31.jdk/Contents/Home/bin/java (found suitable version "1.8.0.31", minimum required is "1.6")
-- Detecting MATLAB JVM version
CMake Warning at cmake/config.cmake:112 (message):
  MATLAB JVM version (1.7) does not match installed Java Development Kit
  version (1.8)
Call Stack (most recent call first):
  cmake/config.cmake:159 (drake_setup_java_for_matlab)
  cmake/config.cmake:255 (drake_setup_java)
  CMakeLists.txt:13 (drake_setup_platform)

-- Found PythonInterp: /usr/local/bin/python2.7 (found suitable version "2.7.12", minimum required is "2.7")
-- Found PythonLibs: /usr/local/Cellar/python/2.7.12/Frameworks/Python.framework/Versions/2.7/lib/libpython2.7.dylib (found suitable version "2.7.12", minimum required is "2.7")
-- Found Python: /usr/local/bin/python2.7 (found suitable version "2.7.12", minimum required is "2.7")
-- Preparing to build eigen
-- Preparing to build spdlog
-- Preparing to build swigmake
-- Preparing to build bullet
-- Preparing to build ccd
-- Preparing to build gflags
-- Preparing to build googletest
-- Preparing to build google_styleguide
-- Preparing to build nlopt
-- Preparing to build sedumi
-- Preparing to build spotless
-- Preparing to build octomap
-- Preparing to build swig_matlab
-- Preparing to build yaml_cpp
-- Preparing to build lcm
-- Preparing to build libbot with dependencies lcm
-- Preparing to build bot_core_lcmtypes with dependencies lcm;libbot
-- Preparing to build robotlocomotion_lcmtypes with dependencies bot_core_lcmtypes;lcm
-- Preparing to build drake with dependencies bot_core_lcmtypes;bullet;ccd;eigen;gflags;google_styleguide;googletest;lcm;libbot;nlopt;octomap;robotlocomotion_lcmtypes;sedumi;spdlog;spotless;swig_matlab;swigmake;yaml_cpp
-- Preparing to build director with dependencies bot_core_lcmtypes;drake;lcm;libbot
-- Preparing to build hsrb_description
-- Configuring done
-- Generating done
-- Build files have been written to: /Users/russt/drake-distro/build_test
30-27-187% cd ../xcode
30-27-187% cmake -GXcode ..
-- CMAKE_INSTALL_PREFIX=/Users/russt/drake-distro/xcode/install
-- The Fortran compiler identification is unknown
CMake Error at cmake/config.cmake:187 (enable_language):
  No CMAKE_Fortran_COMPILER could be found.

Call Stack (most recent call first):
  cmake/config.cmake:253 (drake_setup_fortran)
  CMakeLists.txt:13 (drake_setup_platform)

-- Configuring incomplete, errors occurred!
See also "/Users/russt/drake-distro/xcode/CMakeFiles/CMakeOutput.log".
See also "/Users/russt/drake-distro/xcode/CMakeFiles/CMakeError.log”.
```

jamiesnape, 2016-11-28:
Can you run the `fortran-test` project from before with `cmake -GXcode`? I am seeing the following:
```
$ cat CMakeLists.txt 
cmake_minimum_required(VERSION 3.5)
project(fortran-test LANGUAGES Fortran)

$ mkdir build && cd build
$ cmake -GXcode ..
-- The Fortran compiler identification is unknown
CMake Error at CMakeLists.txt:2 (project):
  No CMAKE_Fortran_COMPILER could be found.

-- Configuring incomplete, errors occurred!
See also "~/fortran-test/build/CMakeFiles/CMakeOutput.log".
See also "~/fortran-test/build/CMakeFiles/CMakeError.log".

$ cat ~/fortran-test/build/CMakeFiles/CMakeOutput.log
The system is: Darwin - 16.1.0 - x86_64
Compiling the Fortran compiler identification source file "CMakeFortranCompilerId.F" succeeded.
Compiler:  
Build flags: 
Id flags: 

The output was:
0
=== BUILD TARGET CompilerIdFortran OF PROJECT CompilerIdFortran WITH THE DEFAULT CONFIGURATION (Debug) ===

Check dependencies
warning: no rule to process file '~/fortran-test/build/CMakeFiles/3.7.0/CompilerIdFortran/CMakeFortranCompilerId.F' of type sourcecode.fortran.f90 for architecture x86_64

Write auxiliary files
write-file ~/fortran-test/build/CMakeFiles/3.7.0/CompilerIdFortran/CompilerIdFortran.build/Debug/CompilerIdFortran.build/Script-2C8FEB8E15DC1A1A00E56A5D.sh
chmod 0755 ~/fortran-test/build/CMakeFiles/3.7.0/CompilerIdFortran/CompilerIdFortran.build/Debug/CompilerIdFortran.build/Script-2C8FEB8E15DC1A1A00E56A5D.sh

PhaseScriptExecution Run\ Script CompilerIdFortran.build/Debug/CompilerIdFortran.build/Script-2C8FEB8E15DC1A1A00E56A5D.sh
    cd ~/fortran-test/build/CMakeFiles/3.7.0/CompilerIdFortran
    /bin/sh -c ~/fortran-test/build/CMakeFiles/3.7.0/CompilerIdFortran/CompilerIdFortran.build/Debug/CompilerIdFortran.build/Script-2C8FEB8E15DC1A1A00E56A5D.sh
GCC_VERSION=com.apple.compilers.llvm.clang.1_0

** BUILD SUCCEEDED **

$ cat ~/fortran-test/build/CMakeFiles/CMakeError.log
Compilation of the Fortran compiler identification source "CMakeFortranCompilerId.F" did not produce an executable in "~/fortran-test/build/CMakeFiles/3.7.0/CompilerIdFortran".
```

jamiesnape, 2016-11-28:
@RussTedrake Can you try #4273 and let me know if it helps?

RussTedrake, 2016-11-28:
The two line CMakeLists.txt also fails with -GXcode:

```
30-27-187% cmake -GXcode .
-- The Fortran compiler identification is unknown
CMake Error at CMakeLists.txt:2 (project):
  No CMAKE_Fortran_COMPILER could be found.

-- Configuring incomplete, errors occurred!
See also "/Users/russt/tmp/fortran/CMakeFiles/CMakeOutput.log".
See also "/Users/russt/tmp/fortran/CMakeFiles/CMakeError.log".
```

that's the same error I was seeing in drake.

david-german-tri, 2016-12-15:
@RussTedrake Is this fixed?

david-german-tri, 2017-01-05:
Pretty sure this is fixed, since I haven't heard ongoing complaints about it.  Closing.

closed 2017-01-05 by david-german-tri