RobotLocomotion/drake #4263: Xcode generator produces warnings on every linking unit due to a bad library include path

RussTedrake, 2016-11-27:
e.g.

```
Ld /Users/russt/drake-distro/xcode/drake/lib/libdrakeMath.dylib normal x86_64
    cd /Users/russt/drake-distro/drake
    export MACOSX_DEPLOYMENT_TARGET=10.11
    /Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/clang++ -arch x86_64 -dynamiclib -isysroot /Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX10.12.sdk -L/Users/russt/drake-distro/xcode/drake/lib -L/Users/russt/drake-distro/xcode/drake/lib/Debug -L/Users/russt/drake-distro/xcode/install/lib/Debug -L/Users/russt/drake-distro/xcode/install/lib -F/Users/russt/drake-distro/xcode/drake/lib -filelist /Users/russt/drake-distro/xcode/drake/math/drake.build/Debug/drakeMath.build/Objects-normal/x86_64/drakeMath.LinkFileList -install_name @rpath/libdrakeMath.dylib -Xlinker -rpath -Xlinker /Users/russt/drake-distro/xcode/drake/lib -Xlinker -rpath -Xlinker /Users/russt/drake-distro/xcode/install/lib -mmacosx-version-min=10.11 -Xlinker -object_path_lto -Xlinker /Users/russt/drake-distro/xcode/drake/math/drake.build/Debug/drakeMath.build/Objects-normal/x86_64/drakeMath_lto.o -Xlinker -no_deduplicate -Wl,-undefined -Wl,error -dynamiclib -Wl,-headerpad_max_install_names /Users/russt/drake-distro/xcode/drake/lib/libdrakeCommon.dylib /Users/russt/drake-distro/xcode/install/lib/libgflags.2.2.0.dylib -single_module -Xlinker -dependency_info -Xlinker /Users/russt/drake-distro/xcode/drake/math/drake.build/Debug/drakeMath.build/Objects-normal/x86_64/drakeMath_dependency_info.dat -o /Users/russt/drake-distro/xcode/drake/lib/libdrakeMath.dylib

ld: warning: directory not found for option '-L/Users/russt/drake-distro/xcode/drake/lib/Debug'
ld: warning: directory not found for option '-L/Users/russt/drake-distro/xcode/install/lib/Debug'
```

Indeed, there is no `Debug` subdirectory in `install/lib`.   This was generated by running `cmake -GXcode -DDISABLE_FORTRAN ..`.  

I don't see the same warning when running the standard makefile generator.

jamiesnape, 2016-11-28:
If I recall correctly, this happens with all projects generated by CMake for Xcode. I will look into it.

jamiesnape, 2016-11-28:
From @bradking and @billhoffman, the Xcode generator adds the `-L` flag because unfortunately we have no way to know which `link_directories` are part of the build and which are external. #4245 may help.

david-german-tri, 2017-04-18:
Closing as wontfix since xcode is not supported and Russ isn't using it anymore.

closed 2017-04-18 by david-german-tri