RobotLocomotion/drake #4287: HybridTrajectoryOptimization fails for systems with zero inputs

mshalm, 2016-11-29:
Recently, when attempting to find limit cycles for a passive, hybrid dynamical system, I cam across an issue with HybridTrajectoryOptimization.m: HybridTrajectoryOptimization.solveTraj assumes that for all modes, the number of inputs is at least one. If you have modes with 0 inputs, you'll get the following error, which is due to the lines starting at 311:

```
Error using mkpp (line 61)
The requested number of polynomial pieces, 14, is incompatible with the proposed`
size, [0], of a coefficient and the number, 0, of scalar coefficients provided.

Error in foh (line 15)
ypp = mkpp(t0,coefs,D);

Error in DircolTrajectoryOptimization/reconstructInputTrajectory (line 116)
      utraj = PPTrajectory(foh(t,u));

Error in HybridTrajectoryOptimization/solveTraj (line 311)
        utraj{i} =
        obj.mode_opt{i}.reconstructInputTrajectory(z(obj.var_offset(i)+1:obj.var_offset(i)+obj.mode_opt{i}.num_vars));
```

hongkai-dai, 2016-11-29:
As a hack, could you make your system to accept one input, and add a constraint that this input is zero? I am afraid we are not going to fix this MATLAB bug any more, but we are implementing a C++ trajectory optimization framework.

closed 2016-11-29 by hongkai-dai

mshalm, 2016-11-29:
I made a (somewhat sloppy) edit to HybridTrajectoryOptimization.m for personal use so it's no issue to me.