RobotLocomotion/drake #4291: RigidBody::body_index_ is not a good unique identifier in RBT

amcastro-tri, 2016-11-29:
`RigidBody::body_index_` changes with calls to `compile()` or more specifically `RBT::SortTree()`. We need a unique identifier for `RigidBody`'s to, for instance, reference them properly in the `KinematicsCache` without creating undesired circular dependencies ([see comment here](https://github.com/RobotLocomotion/drake/pull/4278#issuecomment-263452646)).
Also we should be able to guarantee the user with a unique identifier for each `RigidBody` that does not change during the lifespan of `RigidBodyTree`.

cc'ing @jwnimmer-tri, @liangfok, @sherm1

liangfok, 2016-11-29:
I can also imagine that the body indices will change when we add and remove parts of a tree (like when we tile in and out different parts of the world based on the player character's location).

I suggest adding a `body_id` that's created and applied in a manner similar to `model_instance_id`.

amcastro-tri, 2016-11-29:
Implementation detail body indices will change, yes. However unique body id's will not. The PR I am working on right now introduces `RigidBodyId` (simply a typedef to int) and `RigidBody::get_id()`.
That change alone is helping me removing dependencies in the cache and also helping with the template madness in RBT.

jwnimmer-tri, 2016-11-29:
FWIW, I think semantic typedefs to PODs are really hard to maintain.  If it's just an `int`, either call it an `int`, or make it a `struct BodyId { int value; }` so that the typechecker helps you.  Having types that look like the typechecker will help but don't are frustrating for maintainers.

sherm1, 2016-11-29:
I agree with @jwnimmer-tri that type-safe integers should be used rather than plain typedefs. I would also distinguish general Ids ("handles") from those that are suitable for use as indexes (i.e. they are guaranteed to be small integers in a range 0..n-1 or whatever).

If we use them, type safe integers should be standardized since we are likely to need lots of them. Ideally they have an implicit conversion to int, but no implicit conversion *from* int. I've used macros to create such classes in the past, but I *think* a CRTP implementation could be made to work.

closed 2017-02-01 by amcastro-tri