RobotLocomotion/drake #4294: Rigid body tests fail on minimal build for rigid bodies without geometries

edrumwri, 2016-11-29:
Getting test failure message `unusable_model.cc:9 in updateModel(): assertion 'Compile Drake with a collision library backend for collision support!' failed.` in runge_kutta3_integrator_test.cc on minimal build _even though that test uses a rigid body without geometry_.

Per @SeanCurtis-TRI, there is currently no means for RigidBodyPlant to query whether it contains any geometry (which could be used to short-circuit this problem).

SeanCurtis-TRI, 2016-11-29:
Problem
======
The problem here is a chain of events: 

   1) Minimal builds exclude bullet.
   2) Exclusion of bullet means the `RigidBodyTree` has an `UnusableModel` for its collision model.  All invocations of this model invoke `DRAKE_ABORT`.
   3) Computing the time derivatives of a `RigidBodyPlant` will compute contact forces which, in turn, invoke a method on the `UnusableModel`. 

So, even if you have a test scenario which you know can't have contacts (because you have no collision elements), you can't run that test without bullet being included.  Thus, these tests which logically *could* run in a minimal build can't be included in practice.

Solution
======
One possible solution is for the `RigidBodyPlant` to query contacts *conditionally*.  For example, if no collision elements have been added, no "contact" is possible and `RigidBodyPlant` can infer that the contact forces are zero.  This would require extending the `DrakeCollision::Model` interface to report number of collision elements.  The `UnusableModel` version would always return zero.

Pros:
   - Simple short-term solution to allow such tests to be run in minimal build configurations.
Cons:
   - This is an incomplete solution.  It ignores the case where there is only a single model that *has* collision elements but disallows self-collision.  A count of collision elements would be non-zero, but still no collision would be possible.  We wouldn't want to have two different versions of a robot: one with no collision elements and one with collision elements (which nevertheless disallows self collision) just so we could sometimes run one way, sometimes another.

So, if we choose to go the route of giving the RBP more insight into the underlying data structures and doing something intelligent, we'll have to be careful about how we do this.  Furthermore, adding this functionality to facilitate testing will affect *every* evaluation of time derivatives for every simulation we ever run.  This doesn't feel particularly satisfying.

We should probably find a solution that doesn't punish the normal case just to satisfy this boundary case.  Alternatives:
   1) Modify `UnusableModel` so that instead of throwing errors, it provides "zero" answers.  On the negative side, `UnusableModel` would no longer serve as a hard line which reports that there is no collision detection available.
   2) Leave `UnusableModel` alone and add `NullModel`, which provides zero answers which can be invoked through a test-only compiler flag.  The down side of this is that tests have to be carefully classified to know whether they require a legitimate collision model or if a null model is sufficient.

jwnimmer-tri, 2016-11-29:
Another option would be to add some collision engine (whether bullet, hand-written, xor FCL, whatever) into the "minimal" configuration.

jwnimmer-tri, 2018-12-21:
Bullet is no longer optional (as of a _long_ time ago), so this issue is moot.

closed 2018-12-21 by jwnimmer-tri