RobotLocomotion/drake #4302: `RigidBody::collision_element_groups_` zombie code to be removed

SeanCurtis-TRI, 2016-11-29:
I've got some code in C++ that appears to *not* be used and is cluttering up the place.  

Specifically, it is the concept of "collision element groups".  It is important to note these groups are *not* the collision filter groups defined in a URDF file.  These are a *different* set of groups.

The documentation for the underlying member is as follows:

> // A map of groups of collision element IDs. This is just for conveniently
> // accessing particular groups of collision elements. The groups do not imply
> // anything in terms of how the collision elements relate to each other.
> std::map<std::string, std::vector<DrakeCollision::ElementId>>
>     collision_element_groups_;

This map is almost *unused* in the C++ code.  There is on example that seems to imply how they were intended to be used: `RigidBodyTree::getTerrainContactPoints`.  Each body defines a set of "terrain contact points" based on its collision elements.  By default, all of a body's collision elements *can* contribute to the set.  However, there is an additional argument that allows me to limit the effect to a specific group.  For example, if my foot has a bunch of collision elements and some belong to the group "toe", some "heel", and others "junk", I could get the terrain contact points for just the "toe" elements.  However, this group selection functionality is *not* used.  In all usages that I could find, we are always using all of the collision elements.

It appears we have no path in the C++ code that actually exercises this. 

So, this is a shout out to find out if anyone knows something more specific about these before I kill them.  

P.S. This *type* of logic is used in matlab.  However, the matlab code has its own `RigidBody.m` which defines this functionality and appears to be unconnected to the C++ code.

@RussTedrake @avalenzu @amcastro-tri

patmarion, 2016-11-30:
Yes, this API is used for getting named contact points, for example the toe and heel points of our walking robots.  We store contact point information in URDF files by defining zero radius spheres under named collision groups.  Perhaps that's a weird convention, but that's how we do it right now :)

There are downstream projects using this API from C++ and Python.  There is also Matlab code, but perhaps that is not using the same C++ code path you are referring to, as Matlab has its own parsers.  Note, many Drake Matlab users (openhumanoids) are using older versions of Drake, so I'm not sure to what extent the Matlab support is still used.

If you'd like to change the C++ API then I can help update downstream code, but the functionality is important to keep around, thanks!

jwnimmer-tri, 2016-11-30:
@patmarion I can see that this part of the RBT API is exported through the pydrake rbtree.i wrapper, but I didn't immediately find any RobotLocomotion search hits for python code that is calling it.  Do you have a link or two with sample call sites?

patmarion, 2016-11-30:
Here's a C++ example in the Atlas fall detector:

https://github.com/openhumanoids/oh-distro/blob/master/software/control/src/AtlasFallDetector.cpp#L161

And here's a Python example in the pydrake IK server:

https://github.com/RobotLocomotion/director/blob/master/src/python/director/pydrakeik.py#L348

Both of these call sites are covered by the openhumanoids dashboard, but note that openhumanoids does not use drake/master.  The pydrake IK server is used on drake/master for kuka iiwa and valkyrie development.

jwnimmer-tri, 2016-11-30:
Yup, I see it in the IK server now at least.  Thanks!

sherm1, 2016-11-30:
I'm unclear about the implications of these uses. Are there any drake repo uses or unit tests for this code? RigidBodyTree and collision handling are undergoing major remodeling -- this code is unlikely to survive unscathed. What should be done with it?

patmarion, 2016-11-30:
I would propose that you keep the functionality, as it is pretty important and widely used.  If you come up with a better way to provide the functionality, I think downstream user can adapt to the new API.

For most of Drake's history Matlab was the primary interface.  You'll find tons of usage of `getTerrainContactPoints` in Matlab in the drake repo.  But it sounds like most or all of the uses of the equivalent C++ method are by downstream projects and not in the drake repo.  If you're worried you might break those callers, maybe add a test before beginning the refactoring?

By the way, I think the word 'Terrain' could be dropped from the API, it's probably just a historical artifact.  It's a method to get named contact points that are stored in the URDF, and originally we only thought about the foot/terrain contact points.

I did not design this API and don't consider myself an owner.  Others can probably speak better about this API, thanks!

SeanCurtis-TRI, 2016-11-30:
This has been helpful.  In the PR I'm currently working on, I'll preserve the current functionality.  However, a better insight into its current/historical usage will allow us to make better decisions going forward.

   1) When this interface is used (i.e., grab collision elements by group name), what does the calling code do with those elements?  
   2) Certainly, the C++ doesn't actually *use* the group names to create those "terrain" contact points.  Does this reflect an incomplete usage of group names on the C++ side?  Are there instances in matlab/python where the "terrain" contact points are explicitly defined w.r.t. group names?
   3) There are instances in matlab where a caller removes all groups except for an enumerated set (see `removeCollisionGroupsExcept` in `RigidBody.m`); what is the effect of that?  To what end is this done?

SeanCurtis-TRI, 2016-12-03:
NOTE: This was introduced due to issue #3732 and added with PR #3830.

patmarion, 2016-12-03:
#3830 made the Python API match the Matlab API.  The `getTerrainContactPoints` function has been in C++ since #901 perhaps?  And the `RigidBody::collision_element_groups_` member has been around even longer, probably since the first implementation of collision elements in c++.

> When this interface is used (i.e., grab collision elements by group name), what does the calling code do with those elements?

For example, I may ask for the heel and toe contact points for each foot in order to determine the shape of the support polygon of the robot.  A common use case is for initializing the quasistatic posture constraint.  Sometimes, rather than heel and toe, we want midfoot and toe, if the heel is overhanging the terrain, like on stairs.

> Certainly, the C++ doesn't actually use the group names to create those "terrain" contact points. Does this reflect an incomplete usage of group names on the C++ side?

I'm not sure I understand the question.  The group names are read from URDF and stored in the `RigidBody::collision_element_groups_` member.

> Are there instances in matlab/python where the "terrain" contact points are explicitly defined w.r.t. group names?

Yes, by convention we use 'heel' and 'toe' for foot contacts, but we have used other names like 'pelvis' as well.  It is application specific.  I think if you grep around you will find lots of examples in Matlab in drake and in openhumanoids.

> There are instances in matlab where a caller removes all groups except for an enumerated set (see removeCollisionGroupsExcept in RigidBody.m); what is the effect of that? To what end is this done?

I suppose that is to speed up simulation, by only testing for collision between the foot contact points and the ground.  Someone else may have a better answer than me.

rdeits, 2016-12-03:
@patmarion yes, you're exactly right about removing the other collision groups being done to speed up simulation.

SeanCurtis-TRI, 2016-12-05:
> 'm not sure I understand the question. The group names are read from URDF and stored in the RigidBody::collision_element_groups_ member.

When a `RigidBodyTree` is "finalized" (via calling the `compile` method), one of the actions it takes is to compute and cache the "contact points" per body.  It does this by calling the so-called `getTerrainContactPoints` method.  However, it does so without passing any specific parameter for "group name", which means the body's contact points are initialized without regard to group names.  It considers *all* collision elements that would produce points, regardless of group.  (Admittedly, that is limited to *very* small spheres and boxes.)  So, my question is related to *this* usage.  Are these cached contact points used?  I get the impression that they probably aren't.  At the very least, your use case suggests that they aren't.

patmarion, 2016-12-05:
I'm not sure the history of it.  It seems possible that those points used to be assumed to be terrain contact points, and then someone added the concept of groups, and now those points include non-terrain contact points and old code is making a false assumption.  I just asked @manuelli and he thinks they do not use the get_contact_points() method in the controller right now.

SeanCurtis-TRI, 2019-03-25:
Closed as being uniquely RBT. If this proves to be useful, this can be re-opened.

closed 2019-03-25 by SeanCurtis-TRI