RobotLocomotion/drake #4318: signed/unsigned error caught in xcode (but not in gnu make build!)

RussTedrake, 2016-11-30:
OK.  I think the error is real (and easy to fix).  

What's got me confused is why on the same machine, same checkout, my gnu make build is ok, but my xcode build is failing.  Something in the sea of compiler flags is different.

Why didn't the build server trip on this?

```
CompileC /Users/russt/drake-distro/xcode/drake/lcm/test/drake.build/Debug/drake_mock_lcm_test.build/Objects-normal/x86_64/drake_mock_lcm_test.o lcm/test/drake_mock_lcm_test.cc normal x86_64 c++ com.apple.compilers.llvm.clang.1_0.compiler
    cd /Users/russt/drake-distro/drake
    export LANG=en_US.US-ASCII
    /Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/clang -x c++ -arch x86_64 -fmessage-length=0 -fdiagnostics-show-note-include-stack -fmacro-backtrace-limit=0 -Wno-trigraphs -fpascal-strings -O0 -Wno-missing-field-initializers -Wno-missing-prototypes -Wno-return-type -Wno-non-virtual-dtor -Wno-overloaded-virtual -Wno-exit-time-destructors -Wno-missing-braces -Wparentheses -Wswitch -Wno-unused-function -Wno-unused-label -Wno-unused-parameter -Wno-unused-variable -Wunused-value -Wno-empty-body -Wno-uninitialized -Wno-unknown-pragmas -Wno-shadow -Wno-four-char-constants -Wno-conversion -Wno-constant-conversion -Wno-int-conversion -Wno-bool-conversion -Wno-enum-conversion -Wno-shorten-64-to-32 -Wno-newline-eof -Wno-c++11-extensions -DCMAKE_INTDIR=\"Debug\" -DEIGEN_MPL2_ONLY -DHAVE_SPDLOG -isysroot /Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX10.12.sdk -fasm-blocks -fstrict-aliasing -Wdeprecated-declarations -Winvalid-offsetof -mmacosx-version-min=10.11 -g -Wno-sign-conversion -Wno-infinite-recursion -Wno-move -I/Users/russt/drake-distro/xcode/drake/lcm/test/Debug/include -I/Users/russt/drake-distro/drake/.. -I/Users/russt/drake-distro/xcode/install/include/eigen3 -I/Users/russt/drake-distro/drake/thirdParty/bsd/spruce/include -I/Users/russt/drake-distro/xcode/install/include -I/Users/russt/drake-distro/xcode/drake/lcmtypes -I/Users/russt/drake-distro/xcode/drake/lcm/test/drake.build/Debug/drake_mock_lcm_test.build/DerivedSources/x86_64 -I/Users/russt/drake-distro/xcode/drake/lcm/test/drake.build/Debug/drake_mock_lcm_test.build/DerivedSources -Wmost -Wno-four-char-constants -Wno-unknown-pragmas -F/Users/russt/drake-distro/xcode/drake/lcm/test/Debug -Werror=all -Werror=ignored-qualifiers -Werror=overloaded-virtual -Werror=shadow -Werror=inconsistent-missing-override -Werror=sign-compare -DGTEST_DONT_DEFINE_FAIL=1 -DGTEST_DONT_DEFINE_SUCCEED=1 -DGTEST_DONT_DEFINE_TEST=1 -fPIE -std=gnu++14 -MMD -MT dependencies -MF /Users/russt/drake-distro/xcode/drake/lcm/test/drake.build/Debug/drake_mock_lcm_test.build/Objects-normal/x86_64/drake_mock_lcm_test.d --serialize-diagnostics /Users/russt/drake-distro/xcode/drake/lcm/test/drake.build/Debug/drake_mock_lcm_test.build/Objects-normal/x86_64/drake_mock_lcm_test.dia -c /Users/russt/drake-distro/drake/lcm/test/drake_mock_lcm_test.cc -o /Users/russt/drake-distro/xcode/drake/lcm/test/drake.build/Debug/drake_mock_lcm_test.build/Objects-normal/x86_64/drake_mock_lcm_test.o

In file included from /Users/russt/drake-distro/drake/lcm/test/drake_mock_lcm_test.cc:5:
/Users/russt/drake-distro/xcode/install/include/gtest/gtest.h:1392:11: error: comparison of integers of different signs: 'const int' and 'const unsigned long' [-Werror,-Wsign-compare]
  if (lhs == rhs) {
      ~~~ ^  ~~~
/Users/russt/drake-distro/xcode/install/include/gtest/gtest.h:1421:12: note: in instantiation of function template specialization 'testing::internal::CmpHelperEQ<int, unsigned long>' requested here
    return CmpHelperEQ(lhs_expression, rhs_expression, lhs, rhs);
           ^
/Users/russt/drake-distro/drake/lcm/test/drake_mock_lcm_test.cc:33:3: note: in instantiation of function template specialization 'testing::internal::EqHelper<false>::Compare<int, unsigned long>' requested here
  EXPECT_EQ(message_size, published_message_bytes.size());
  ^
In file included from /Users/russt/drake-distro/drake/lcm/test/drake_mock_lcm_test.cc:5:
/Users/russt/drake-distro/xcode/install/include/gtest/gtest.h:1924:63: note: expanded from macro 'EXPECT_EQ'
                      EqHelper<GTEST_IS_NULL_LITERAL_(val1)>::Compare, \
                                                              ^
1 error generated.
```

jwnimmer-tri, 2016-11-30:
Is `ccache` or `CCACHE_CPP2` involved?  As it turns out, clang warnings are different depending on whether the source code has been pre-processed yet.

We have also seen where @mwoehlke-kitware has signed-unsigned warnings that we don't; we suspected some kind `-I` vs `-isystem` difference perhaps?  I don't recall if we ever root-caused that one.

jamiesnape, 2016-11-30:
IIRC, Xcode also has its own set of default flags.

jamiesnape, 2016-11-30:
Also, you are using a newer compiler, if the configuration of the machine is the same as the one in #4259.

mwoehlke-kitware, 2016-11-30:
Yeah, that looks almost *exactly* like the sorts of errors I've seen (where *only* I see them). Possible data point: I do use ccache.

jwnimmer-tri, 2016-11-30:
@mwoehlke-kitware Do you have `CCACHE_CPP2=1` set in your environment (you should).

jamiesnape, 2016-11-30:
The compiler is `/Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/clang`.

jamiesnape, 2016-11-30:
Not sure Xcode would use `ccache` in that case.

mwoehlke-kitware, 2016-11-30:
> Do you have `CCACHE_CPP2=1` set in your environment (you should).

No. According to the manpage it disables an optimization and "is primarily a debugging option". Can you explain why you believe I should have it set?

jamiesnape, 2016-11-30:
You tend to get errors about unexpanded macros and other craziness when using `clang` and `ccache` together otherwise.

jonathan-owens, 2017-01-20:
This is continuing to happen: https://drake-cdash.csail.mit.edu/viewBuildError.php?buildid=322732

jonathan-owens, 2017-02-16:
https://drake-cdash.csail.mit.edu/viewBuildError.php?buildid=394069

jamiesnape, 2017-02-16:
Yes, we have lots. They are on the backlog to fix.

david-german-tri, 2017-04-11:
Since no one is actually using xcode anymore TTBOMK, I'm going to close this as wontfix.  Signed/unsigned comparison in test assertions seems like too unimportant a defect to spend time boilerplating.  (It's hard to imagine anyone writing a test that accidentally passes because of integer underflow.)

closed 2017-04-11 by david-german-tri