RobotLocomotion/drake #4325: infinite loop simulating discrete systems

RussTedrake, 2016-12-01:
Here's a fun one.  
Try evaluating LeafSystem's GetNextSampleTime: https://github.com/RobotLocomotion/drake/blob/master/drake/systems/framework/leaf_system.h#L302

with a period set to 0.015.  At time ~0.165, time stops.  :)  why?
that's timestep 11.  but due to floating point fun, we get:
```
prev_k = floor(0.015*11/0.015) = 10
next_k = ceil(0.015*11/0.015) = 11
```
since `prev_k != next_k`, therefore `next_k = 11`.  Which is the same time that we started.  And round and round it goes.

RussTedrake, 2016-12-01:
fwiw... changing to
`const int64_t prev_k = static_cast<int64_t>(floor((offset_time + std::numeric_limits<T>::epsilon())/ period));` resolves it, but I didn't think long about what the right solution is.

RussTedrake, 2016-12-01:
there are some more haunts in there, too.  with that fixed and using a period of 0.001, time keeps moving forward, but my update method gets called less and less frequently:
```
t = 0.001
t = 0.002
t = 0.003
t = 0.004
t = 0.005
t = 0.006
t = 0.007
t = 0.008
t = 0.01
t = 0.011
t = 0.014
t = 0.015
t = 0.019
t = 0.022
t = 0.027
t = 0.03
t = 0.037
t = 0.044
t = 0.053
t = 0.06
t = 0.073
t = 0.088
t = 0.105
t = 0.12
t = 0.145
t = 0.176
t = 0.209
t = 0.24
t = 0.289
```

RussTedrake, 2016-12-01:
fwiw -- simulating with period .0025 or 0.003 for 3 seconds still gets stuck in an infinite loop using my quick fix.

hongkai-dai, 2016-12-01:
Should the epsilon be multiplied by T?

edrumwri, 2016-12-01:
Oooh! A floating point challenge- I accept.

sherm1, 2016-12-01:
Crap. I think I'm to blame for that code. I believe it got promoted from a hack in a test case I wrote.

sherm1, 2016-12-01:
Using epsilons makes me nervous as being vaguely nondeterministic. Better if the algorithm can be expressed with only large numbers. How about:
```c++
int64_t prev_k = static_cast<int64_t>(floor(offset_time / period)); // unchanged

const T sliver(period/16); // Presumably >> epsilon.
if (prev_k*period < offset_time - sliver)
  ++prev_k; // floor fell to the previous period; now it is current.

// The rest is unchanged, but k gets bumped more often.
const int64_t next_k = static_cast<int64_t>(ceil(offset_time / period));
const int64_t k = (prev_k == next_k) ? next_k + 1 : next_k;
return offset + (k * period);
```
(Sorry, didn't actually try it.)

david-german-tri, 2016-12-01:
I think this is my new favorite Drake bug.  I'm not totally convinced by @sherm1's approach above, since the integrator could legitimately stop within a sliver of a sampling moment, and we wouldn't want to just skip that sample...

I will mull it more in the hopes of PRing a candidate fix before the PST crowd wakes up.

david-german-tri, 2016-12-01:
(Also, this is a prime candidate for TDD.)

edrumwri, 2016-12-01:
I don't know what TDD is.

Russ, what floating point value is returned by the function in this case?
Is it zero? A (small) negative number?

david-german-tri, 2016-12-01:
> I don't know what TDD is.

Test-driven development (which would probably answer your question to Russ...in the process of writing a repro test case myself right now)

RussTedrake, 2016-12-01:
I believe you can reproduce it by simply changing the sampling time on 
https://github.com/RobotLocomotion/drake/blob/master/drake/examples/simple_discrete_time_system.cc

fwiw -- I caught the missed time samples quickly because I was programming/debugging graphically with the matlab plotting tool.

SeanCurtis-TRI, 2016-12-01:
I'm going to throw something in.  I *think* there's a simple answer, but given that it isn't what is implemented, I would love for someone to tell me what its problem is. I figured I'd crowd source the analysis.  :)

(I apologize for the interface change.  It's a quick and dirty hacked proof-of-concept.  After a simple test, I was able to confirm that, at least in a simple case, the problem was alleviated.  Generally, I'm inclined to believe that this addresses the issue.)

```
double GetNextSampleTime( double offset, double period, double curr_time ) {
  using namespace std;
  double offset_time = curr_time - offset;
  int64_t next_k = static_cast<int64_t>(ceil(offset_time / period));
  double next = offset + (next_k * period);
  while ( next <= curr_time ) { // technically, this should only ever execute at most one iteration.
    ++next_k; 
    next = offset + (next_k * period);
  }
  return next;
}
```

david-german-tri, 2016-12-01:
@SeanCurtis-TRI kudos, that seems right to me.  #4331

SeanCurtis-TRI, 2016-12-01:
Obviously, we can replace the `while` loop with a single if (based on the assertion that the ceil function can never be off by more than a single step.)

david-german-tri, 2016-12-01:
Uh, looks to me like https://github.com/RobotLocomotion/drake/issues/4325#issuecomment-264070247 is another bug unrelated to the OP, somewhere in `systems/analysis`...digging into that now.

david-german-tri, 2016-12-01:
Suspicious: https://github.com/RobotLocomotion/drake/blob/master/drake/systems/analysis/runge_kutta3_integrator-inl.h#L74

Adding some debug instrumentation to `IntegratorBase<T>::StepOnceAtMost`:

```
0.001
0.002
0.003
0.004
0.005
0.006
0.007
0.008
[2016-12-01 12:14:39.745] [console] [info] kTimeHasAdvanced
0.01
0.011
[2016-12-01 12:14:39.745] [console] [info] kTimeHasAdvanced
[2016-12-01 12:14:39.745] [console] [info] kTimeHasAdvanced
0.014
0.015
[2016-12-01 12:14:39.745] [console] [info] kTimeHasAdvanced
[2016-12-01 12:14:39.745] [console] [info] kTimeHasAdvanced
[2016-12-01 12:14:39.745] [console] [info] kTimeHasAdvanced
0.019
```

david-german-tri, 2016-12-01:
Sometimes `max_step_size` is smaller than `update_dt` in `IntegratorBase<T>::StepOnceAtMost`.

david-german-tri, 2016-12-01:
Ah - the default maximum step size in simulator.h is 0.001, so when the update period configured in our system leads to a sampling time that is a multiple of 0.001, there is floating-point instability.

This will happen a lot with a period of 0.001 in particular, but not so much with a period of 0.015.

RussTedrake, 2016-12-04:
simulating with timestep 0.001 for 2 seconds still results in my system being called only twice.

edrumwri, 2016-12-04:
Several of us that worked on issue #4325 were worried that it wasn't going
to address this second problem you experienced. I expected and continue to
expect to encounter further bugs as we continue to stress the simulator.

david-german-tri, 2016-12-05:
> simulating with timestep 0.001 for 2 seconds still results in my system being called only twice.

I can't reproduce this.  Are we still talking about `simple_discrete_time_system.cc` (https://github.com/RobotLocomotion/drake/issues/4325#issuecomment-264213109)?

david-german-tri, 2016-12-12:
I believe we determined this was actually fixed by #4331 and #4372, so I'm closing again.  @RussTedrake feel free to reopen if we've still missed something here.

PR #4331 (2016-12-02): Ensure that the next sample time is always greater than the present time.
Substantive idea here comes from @SeanCurtis-TRI.  Fixes #4325.

---
This change is [[image]](https://reviewable.io/reviews/robotlocomotion/drake/4331)
closed 2016-12-12 by david-german-tri