RobotLocomotion/drake #4326: Add Drake C++ support for ROS Xacro

liangfok, 2016-12-01:
In #4218, the HSRb model is not being used in `hsrb_system_factories_test` because it is unable to generate the URDF model of the HSRb due to lack of a modern version of ROS `xacro`. This issue tracks the updating and integration of the latest version of `ROS xacro` with C++ Drake.

liangfok, 2016-12-01:
An alternative approach is to use the latest version of ROS `xacro` to generate the URDF and save the URDF into a file that's committed to this repository. I'd prefer local on-demand generation rather than version controlling generated code, but this might be an easier short term solution.

liangfok, 2017-04-12:
Based on #5660 and #5636, I think relying on the ROS-installed version of `xacro` to generate the URDF files, and then committing the resulting URDF files into `master` is adequate. To avoid confusing users into thinking the version of `xacro` in Drake is usable (it's too old and does not support features needed by some `.xacro` files in `master`), we should probably remove it from master.

liangfok, 2017-04-13:
Based on feedback from #5807, the new plan is to:

1. Update Drake's version of `xacro` to [a more recent version](https://github.com/ros/xacro/blob/kinetic-devel/src/xacro/__init__.py).
2. Use Bazel's [genrule()](https://bazel.build/versions/master/docs/be/general.html#genrule) to generate the URDFs.

jwnimmer-tri, 2017-04-13:
I'd encourage the approach for (1) to be to pull `xacro` into the `WORKSPACE`, rather than updating the vendored copy in-tree.

jwnimmer-tri, 2020-04-08:
@sammy-tri Could you post a note about recent need for xacro, please?

sammy-tri, 2020-04-08:
In #13007, I managed to get the collision models for the Jaco arm-only and arm + hand models out of sync when manually merging changes between the two.  Building arm only and arm + hand models from the same source is a typical use for xacro, Franka does so for their model of the panda, for example in https://github.com/frankaemika/franka_ros/tree/kinetic-devel/franka_description/robots  Additionally, since the upstream versions of models like the Jaco and Panda use xacro, keeping drake's models in sync with upstream would be simplified.

jwnimmer-tri, 2020-04-08:
https://github.com/jwnimmer-tri/drake/commits/ros_xacro appears to work with `--help` and one of Anzu's xacro files as a sanity check.  Probably just needs a regression test added and we can PR to Drake.

\CC @EricCousineau-TRI @IanTheEngineer FYI

PR #13022 (2020-04-09): workspace: Add ros_xacro
We plan to use this to manage robot URDFs, especially when the upstream vendor of the URDFs already stores them in xacro'd form.

Closes #4326.

---
This change is [[image]](https://reviewable.io/reviews/robotlocomotion/drake/13022)
closed 2020-04-09 by jwnimmer-tri