RobotLocomotion/drake #4327: Standardising manipulation example / demo names

naveenoid, 2016-12-01:
With the KUKA IIWA and Schunk examples getting populated and Toyota HSR on the pipeline, it might make sense to start standardizing on a naming style. 

I propose the following 

/example/brand-name_robot-name_model-name-or-type<if any>/brand-name_robot-name_demo-name

For instance 
/example/kuka_iiwa_arm/kuka_iiwa_passive_control_demo.cc

Opinions?

liangfok, 2016-12-01:
Oops. I just changed all of the HSR demos to be [demo name]_[brand name]_[model_name], per [this comment](https://reviewable.io/reviews/robotlocomotion/drake/4218#-KXoltVB_8WJi_sMDHXd). Either way is OK with me though; I don't strongly favor one over the other.

naveenoid, 2016-12-01:
I think you can proceed with landing that PR..we can easily make a followup PR with renaming once we finish bike-shedding.. Its just worth considering it as we start supporting more and more commercial robot platforms as opposed to the "traditional" drake (brand)nameless example systems.

liangfok, 2016-12-01:
Thinking about this some more, I think I prefer `[brand name]_[model name]_[demo name]`. I'll go ahead and apply this to the HSRb demo.

jwnimmer-tri, 2016-12-01:
Remember that soon you won't have to repeat the directory name in the program name.  So `/examples/kuka_iiwa_arm/passive_control_demo` will never programmatically conflict with `/examples/hsrb/passive_control_demo`.  If you want to add more to the name for developer readability fine, but nothing in the compilation, test, or release process will force you to uniqueify the basenames.

naveenoid, 2016-12-01:
just nice to have some standard example naming styles..

liangfok, 2016-12-01:
Sorry for the constant changes... for the HSRb's ROS demos. I'm now using `drake_examples_[brand name]_[model name]` as the ROS package name. Then, for the actual files within this package that constitute a demo, I'm just using `[demo name]`.

sammy-tri, 2016-12-01:
I'm for drake/examples/[brand_name]_[model_name] (which I think is just agreement with @liangfok) for the directory name.  I'm not sure I like constantly writing "arm" or "gripper" or whatever.  Do we then get drake/examples/valkyrie_humanoid?  (now that I've written that, I'm not sure we should be too rigid about this.  drake/examples/boston_dynamics_atlas_humanoid doesn't exactly seem like a wonderful name)

liangfok, 2016-12-01:
I would recommend `drake/examples/nasa_valkyrie_[version]` and `drake_examples/bd_atlas_[version]` (note acronym abbreviation and optional version indicator).

sammy-tri, 2019-02-14:
Closing since the current set of examples along with our usage in the bazel build doesn't seem to be causing any confusion.

closed 2019-02-14 by sammy-tri