RobotLocomotion/drake #4328: Provide simple API to RBT users

amcastro-tri, 2016-12-01:
This is something that has been requested by several people in the past months. 
Once the work on templating RBT on `<T>` ( #3987) is done, I'd like to provide a simple API like so:

```c++
RigidBodyPlant model("my_model.sdf"); // This is pseudo-code, in reality we'd use a parser.
Context<T> context = model.CreateDefaultContext();
// The user knows that "elbow" is a revolute joint and therefore calls the specific method.
// This method would throw if "elbow" is not a RevoluteJoint.
RevoluteJoint<T>& elbow_joint = model.GetRevoluteJointByName("elbow");
// Now user makes RevoluteJoint specific requests.
T elbow_joint_angle = elbow_joint.get_angle(context);
// And similar API's for RigidBody's
const auto& forearm = model.GetRigidBodyByName("forearm");
Vector3<T> xc = forearm.get_center_of_mass_world(context);
```

Users then don't even need to know the specifics of the context or even the specifics about what generalized coordinates are in use. The user just focus on making physically meaningful questions to the RBT and its components.

amcastro-tri, 2016-12-01:
Notice that in the snippet above I am making the following implicit proposals:
 - Users do not instantiate n `RigidBodyTree` directly but rather instantiate a `RigidBodyPlant` which provides `CreateDefaultContext()`.
- `RigidBodyTree` components such as joints and bodies now would have query methods taking a `Context<T>` by argument (coupling with System 2.0, or actually just its context).

This is issue is therefore created to arrive to a decision on how this API should look like and the specifics of this coupling  with the System 2.0 `Context`.

cc'ing @sherm1, @david-german-tri, @jwnimmer-tri since we talked about this before but now I am getting closer to actually implement it.

david-german-tri, 2016-12-01:
At a high level, this sounds awesome.

> This is issue is therefore created to arrive to a decision on how this API should look like and the specifics of this coupling with the System 2.0 Context.

At a low level, I'd like to avoid that coupling.  Would it suffice to have a `RigidBodyState<T> : public BasicVector<T>` as the `ContinuousState` in the Context, and then do something like this?

```
const RigidBodyState<T>& state = model.get_state(context);
RevoluteJoint<T>& elbow_joint = model.GetRevoluteJointByName("elbow");
T elbow_joint_angle = elbow_joint.get_angle(*state);
```

david-german-tri, 2016-12-01:
Thinking for five more minutes, you probably also need Parameters and some data from the Cache.  So, put more generally: Could we have an accessor on `RigidBodyPlant` that takes a `Context` and returns a struct of const references to just the particular data that's needed to ask questions of RBT components?

```
const RigidBodyFacts<T> facts = model.get_facts(context);
T elbow_joint_angle = elbow_joint.get_angle(facts);
```

I think this indirection is useful to avoid linking RBT components to the System framework, to prevent RBT components from looking at things they shouldn't, like time or inputs, and to avoid requiring RBT components to rummage around in the Cache.

amcastro-tri, 2016-12-01:
> Would it suffice to have a RigidBodyState<T> : public BasicVector<T> as the ContinuousState in the Context, and then do something like this?

No, it wouldn't. A method like `RigidBody::get_center_of_mass(of some sort of context)` needs access to the `Cache<T>`, which could simply be wrapping a `KinematicsCache<T>` on a first cut. It'd look something like:
 ```c++
const VectorX<T>& RigidBody::get_center_of_mass(const Context<T>& context) const {
  const auto& kinematics_cache = context.GetCachedValue(kc_ticket)->GetValue<KinematicsCache<T>>();
  auto& kc_element = kinematics_cache.getElement(*this);// or much better getElement(this->body_unique_id())!!
  const Isommetry3<T>& X_WB = kc_element.transform_to_world;
  return X_WB * get_centr_of_mass(); 
}
```
where `kc_ticket` was intialized by `RBP::CreateDefaultContext()`. That ticket could live in `RBT` and we could query the parent `RBT` for the cache entry, but the above is just a snippet trying to show that we need to access cached computations for such simple requests. Other requests would include asking for velocities, accelerations. 
More complex results than just angles for joints would include for instance querying for joint forces, which would require access to cached Lagrange multipliers and joint Jacobians.

david-german-tri, 2016-12-01:
Makes sense...see my second post above.  https://github.com/RobotLocomotion/drake/issues/4328#issuecomment-264197594

amcastro-tri, 2016-12-01:
Ah! I just see your reply right after I pressed the "Comment" button. Yes, you got it right, I need access to the cache!. Your `RigidBodyFacts<T>` class I think will look much like `Context<T>` and I think it'd be nicer to provide users with a uniform API across Drake. Having users to separately think about `Context<T>` and `RigidBodyFacts<T>` when they are pretty much the same concept seems to complicate things (and a duplication of code).

> I think this indirection is useful to avoid linking RBT components to the System framework, to prevent RBT components from looking at things they shouldn't

That is a responsibility of the developers, not the users.

david-german-tri, 2016-12-01:
> Your RigidBodyFacts<T> class I think will look much like Context<T> and I think it'd be nicer to provide users with a uniform API across Drake.

A critical difference (I think) is that it needn't include the whole Cache.  It would be something like:

```
RigidBodyFacts<T> {
  const RigidBodyState<T>& state;
  const KinematicsCache<T>& kinematics;
}
```

> That is a responsibility of the developers, not the users.

Sure, but perhaps we can more naturally fulfill that responsibility with some encapsulation.

edrumwri, 2016-12-01:
So you would have to have a method for every single type of joint that
might be present in the model? If I add a new joint it would have to be
updated?

amcastro-tri, 2016-12-01:
@edrumwri, are you referring to `GetRevoluteJointByName()`? the answer would be yes. Of course you can let users also do `dynamic_cast<MyJointType>(GetJointByName());`.

amcastro-tri, 2016-12-01:
@david-german-tri. Cleary this needs more discussion (and hence this issue) but another reason to want `Context<T>` (and `Cache<T>`) here is for future improvements to our caching mechanism. Those improvements could make use of our general `Cache<T>` capability while in the `KinematicsCache<T>` world everything is set on stone.

liangfok, 2016-12-01:
Instead of:

```
RigidBodyPlant model("my_model.sdf");
```

In the long term, based on the conversation in #4317, I believe it should take as input some Intermediate Representation:

```
RigidBodyPlant model(intermediate_representation);
```

In the short term, I think it should just take a `unique_ptr<RigidBodyTree<T>>` as an input parameter, and the `RigidBodyTree<T>` can be instantiated using a parser.

david-german-tri, 2016-12-01:
@liangfok - I agree, and I think that was implied in @amcastro-tri's original post when he called the constructor pseudo-code.

Possibly there should be a factory function `MakeRigidBodyPlant("my_model.sdf")` that handles the parse->IR->constructor, but that's a minor detail.

amcastro-tri, 2016-12-01:
Exaclty right @david-german-tri. @liangfok look at the comment right next to that line that says "This is pseudo-code, in reality we'd use a parser.". The question I am trying to answer here is not about parsers but about how exactly to pass `Context`/`Cache` info to multibody code should it be actual System 2.0 objects or some other proxy.

sherm1, 2016-12-01:
Sorry I'm late on this -- I was at the dentist! I do like what @amcastro-tri proposed above a lot. We should plan on using the Context as the source of state and cached info for RBtree, and initially stuff the whole kinematics cache there until we can make better use of the new caching capabilities.

liangfok, 2016-12-01:
Sorry, I did not see the pseudo comment above.

Regarding the subsequent API proposal, it appears to assume that there is only one model instance inside the tree. Let's not forget the need for a `model_instance_id` parameter to distinguish between multiple model instances when more than one instance of the same model are in the tree.

amcastro-tri, 2016-12-02:
The solution to that is having methods (that we do now) like so: 
```c++
const auto& forearm = model.GetRigidBodyByName("forearm", model_instace_id);
```

closed 2017-02-01 by amcastro-tri