RobotLocomotion/drake #4330: bool System<T>::has_undeclared_state()  ?

RussTedrake, 2016-12-01:
We've talked about this a few times, but I think it's time that we should introduce a method / trait / or some other attribute that can be used to declare systems to have limited implementations.  My draft RandomSource has this property.  But other systems will, too (@manuelli asked for it, driving has asked too).  

The implications of having undeclared state are e.g., that (a) we should only use fixed-step integration, (b) logs will only make sense if they are started from time=0, (c) we shouldn't expect linearization or other forms of analysis to work.

Although the most obvious choice would be to default this to true, I would love to make it false and force people to derive from a "BadLeafSystem<T>" or something where the default is true.

@david-german-tri , @jwnimmer-tri , @sherm1 -- what do you think?

manuelli, 2016-12-01:
In particular my use case is as follows. I often need to write a controller that has lots of "state," i.e. `if` statements and the like, popping off queues of footsteps etc. This is kind of logic is hard to put into what we traditionally think of as state in a system. In order for this to work in a simulation context I need to ensure that this controller is called at times that are always increasing and preferably at a fixed rate, i.e. call my controller every dt seconds, and only forwards in time. In particular I don't care what sort of complicated simulation goes on to get the underlying robot system from t ---> t+dt (in other words it's ok to use some variable step solver here that thinks about hybrid guards etc, my controller has already provided a control input for this time interval so the dynamics of the robot are well defined). But my system should only be called at times {t,t+dt, t + 2*dt, . . . . } and so doesn't care what happens in between.

david-german-tri, 2016-12-01:
@manuelli Have you taken a look at https://github.com/RobotLocomotion/drake/blob/master/drake/examples/simple_discrete_time_system.cc?  Combined with abstract state, I think that approach would meet your needs as described without resorting to hidden state.  (All your substantive work would be in the state update, not in `EvalOutput`.)

jwnimmer-tri, 2016-12-01:
I think in my (conceptual) `UndeclaredStateSystem` base class that I use to shell out to third-pary black-box code, I can already do Russ's item (a) by declaring event timing and not declaring any continuously-differentiable state.  So for that, I don't think we need a separate attribute method.

I guess if Russ's item (c) requires something beyond that, we'd need to add a getter.  But wouldn't it be enough to just never even try to run the control analysis on a Diagram that has black-box behavior?  Like, when would someone accidentally try to do that, and do we really need to fail-fast in that case?

manuelli, 2016-12-01:
I couldn't find a header file for abstract_state.h but I am guessing it can be any object/struct. The context contains this abstract state. It is passed in as a const to the the `DoEvalDifferenceUpdates` and then I encode what I want to change in the `updates` pointer, is this correct?

Is it possible to just get a pointer/reference to the Abstract state and be able to modify it myself in the `DoEvalDifferenceUpdates` method rather than encoding the changes in the 'updates' pointer? I say this because some of my use cases require modifying the state during one tick, and several function calls within that tick depend on this modified state.

sherm1, 2016-12-01:
@manuelli, we haven't yet added the method for updating abstract state variables (we're calling those "mode variables"). As soon as we do (probably a few weeks) you will be able to do exactly what you wanted to do above. There will be an update method that is passed a mutable Context so that you can get write access to your mode variables and modify them safely.

sherm1, 2016-12-01:
@RussTedrake, if we provide active support for unsafe systems I'm worried that it will encourage people to use those rather than to think about how to structure their computations using the tools we provide for high-accuracy simulation of hybrid systems; that is more difficult so I think the default will be that people won't do it. I do think it's a good idea to have a runtime method to convey that a system has hidden state if that's how it *has* to be. I would want to distinguish systems that may update their hidden state during `EvalTimeDerivatives()` (preventing variable step integration) from those that have hidden state but promise only to update it between steps (meaning only logging and playback would be problematic). In any case we ought to somehow make this seems like a shameful state of affairs so people won't do it lightly!

RussTedrake, 2017-09-30:
This came up again in a discussion with @travelbureau yesterday.  A few updated thoughts:

It we want to put an arbitrary piece of computation under test in our systems framework (e.g. a library we have only in binary form), then there are multiple ways that it might violate our “well-behaved system” model:
1) black box (e.g. only supports doubles, not other scalar types).  this case is already handled.
2) has_undeclared_state, as discussed above.
3) has_undeclared_randomness.  this can happen in subtle ways (e.g. the system under test is a message passing system that does not intentionally introduce undeclared randomness, but is not guaranteed deterministic).

2 and 3 seem like major disconnects from the system’s framework, to the point where it makes sense to ask if we should still use the system framework at all to analyze them.  But there are still very meaningful queries that we can / should be able to perform on them.  For instance, I’d like to use the declared state to specify my requirements (e.g. cars don’t crash) in the form of constraints, then run eg. monte-carlo simulation techniques to falsify those requirements.  

@sherm1 — yes, setting these flags (if we implement them at all) will have big warning signs saying “you’re about to disable most all of our best tools”.  But the tools that we can implement for this kind of system are still tools that we want to have available to analyze our well-behaved systems, too.

EricCousineau-TRI, 2019-02-06:
This may have cropped up again in #10508, specifically regarding `Publish()` and whether or not we should require everything upstream of `Publish()` be hermetic, even though things downstream (e.g. LCM publishing, Drake Visualizer or RViz state) are not.

sherm1, 2019-02-06:
I'm not clear on what problem this would be solving now. Systems can use arbitrary abstract state in the Context as a best practice. However, if it's difficult to do that there is nothing technically wrong with updating local state as long as 
- it is done only in an event handler (because those are points at which we are committed to the trajectory and won't need to back up), and 
- it doesn't affect computations that _are_ in the Context (because those are cached and won't get invalidated if they can't see the state changes).

RussTedrake, 2019-02-06:
It’s still very important!  What if you are encapsulating some piece of black-box (or simply complicated) code for use in a drake simulation, and that code has some global variables buried deep in its guts.  Or if you just want to throw together a quick gui with sliders for teleop when the slider state is owned by the window manager.  We want to support these systems in diagrams, but should throw is someone tries to prove something about them.

sherm1, 2019-02-06:
The proposal is just to obtain a hint from the user, right? I suspect many users won't know that they have a global variable buried deep in the guts of some library they are using, so won't know to give us the hint. I don't have any objection to asking for that hint if you think it's worthwhile but I'm skeptical that we'll get reliable information in actual use.

sherm1, 2019-02-06:
BTW there is a class of perfectly-harmless state (mutable counters for reporting stats, e.g.) that don't affect the trajectory so should still permit behavioral proofs. It wouldn't be necessary to set the hint in that case.

RussTedrake, 2019-02-07:
I agree that it cannot/will not serve as some bullet-proof guard.  But want to allow a savvy user can protect themselves by (optionally) setting this property when they bring in code that carries state which could change the behavior of the output.