RobotLocomotion/drake #4333: SDF Parser and links named "world".

SeanCurtis-TRI, 2016-12-01:
Problem
======

There appears to be a design inconsistency in the sdf parser.

When parsing a link specification, if the link is named "world", the code throws an exception as seen [here](https://github.com/RobotLocomotion/drake/blob/master/drake/multibody/parser_sdf.cc#L322).  Clearly, drake strongly disallows links named "world".  However, if the link is *not* named "world", a body is instantiated and that is passed into some downstream methods.  

One downstream method is `ParseSdfCollision`.  It accepts the body as an argument.  It tests to see if that body is named "world" and takes an action if it is so named.  However, this is impossible.  The test should never pass, the corresponding code will never execute.  

Solution
======
The code in question (`ParseSdfCollision`) comes with an extensive comment.  Figure out the intent of the code and implement it in some other way.

liangfok, 2016-12-01:
Looking at the aforementioned extensive comment and the two proposed solutions it mentions, I would go with option (2): "Load collision elements on a separate pass after links and joints were already loaded." Maybe this can be done by switching to [OSRF's SDF parser](https://bitbucket.org/osrf/sdformat) and using the [sdf::SDF](http://osrf-distributions.s3.amazonaws.com/sdformat/api/4.0.0/classsdf_1_1SDF.html) object it provides as an in-memory intermediate representation that contains both joints and links.

SeanCurtis-TRI, 2016-12-01:
By a happy coincidence, that is *exactly* what I'm working on now: deferring registration of collision elements with the collision model until the RBT has been otherwise processed.

The short term solution basically leaves the sdf parser alone.  (Although we have related issues with the urdf parser.)  Which is why I'm doing the fix on the RBT side instead of the parsing side.

tri-ltyyu, 2019-03-25:
Please use MBP instead. If it's still an issue, please open new issue.

closed 2019-03-25 by tri-ltyyu