RobotLocomotion/drake #4339: set_minimum_step_size > set_maximum_step_size

RussTedrake, 2016-12-02:
I just lost some time frustrated that the simulator was ignoring my request to set the minimum_step_size.  Turns out the maximum_step_size had already been set (by the defaults, not by me), and was smaller than my minimum_step_size.  This failed silently.

sherm1, 2016-12-02:
Sorry, Russ. The default handling needs some more refinement. It should wait until all the user-requested settings have been made, then set defaults for unspecified parameters based on those. The default max should definitely be larger than a user-set min!

edrumwri, 2016-12-02:
I'm preparing to submit a PR to throw an exception if the maximum step size
is less than the minimum step size when the integrator is initialized.

We've struggled with the issue of setting good defaults for the simulator.
Even with the error controlled integrator, it's not foolproof.

@sherm1- we should use the characteristic time (and other system info) to
help us set the integration parameters to better defaults.

edrumwri, 2016-12-02:
Fixed with PR #4341.

closed 2016-12-02 by edrumwri