RobotLocomotion/drake #4342: Create an Example of Asynchrony Between Controller and Plant

liangfok, 2016-12-02:
# Problem Definition

TTBOMK, as of December 2, 2016, all demonstrations of a controller and plant using System 2.0 enforces synchrony between the controller and the plant by co-locating both within a single `drake::systems::Diagram` that's executed by a `drake::systems::Simulator`. While this is useful for certain types of analysis, debugging, and testing, it does not reflect reality where the plant is a piece of hardware that by definition operates asynchronously from the controller. To better mimic real-world controller / plant interaction, we should create demos where the controller and plant execute asynchronously.

# Suggested Course of Action

Achieve asynchronous controller / plant execution by triggering a control cycle based on the arrival of an LCM message. Modify `drake::lcm::DrakeLcmInterface` to support event subscribers that are notified when an LCM message arrives. Create a new `drake::systems::ControllerExecutor` class that registers itself as an LCM listener and executes one cycle of the controller each time an LCM message is received. The controller executed by the `drake::systems::ControllerExecutor` communicates over LCM and a local loop-back network to a plant that's being simulated by a `drake::systems::Simulator`. Both the `ControllerExecutor` and `Simulator` can still reside within a single process.

RussTedrake, 2016-12-02:
fwiw - I'm working on examples which run the plant in one exe and the control in another (starting with the acrobot).  but i agree it would be good to figure out how to add these to the testing framework.

jwnimmer-tri, 2016-12-22:
FYI this arose our of discussions with @siyuanfeng-tri.  Assigning to @RussTedrake for prioritization / OKR assignment.