RobotLocomotion/drake #4343: Refine DrakeVisualizer To Support Model Instance-Level Updates

liangfok, 2016-12-02:
# Problem Definition

Currently, `drake::systems::DrakeVisualizer` loads and updates the Drake Visualizer at the granularity of the _entire_ `RigidBodyTree`. As of https://github.com/RobotLocomotion/director/pull/393, Drake Visualizer can how update at a granularity of an individual model instance within a `RigidBodyTree`. As we scale up our simulations with increasing numbers of model instances, the benefits of being able to load and update individual model instances rather than the entire `RigidBodyTree` will increase. Thus, we should consider refining `drake::systems::DrakeVisualizer` to support updates at a model instance granularity.

# Suggested Course of Action

Make use of the new `DRAKE_VIEWER_ADD_ROBOT` channel in place of the current `DRAKE_VIEWER_LOAD_ROBOT`. Modify the current usage of `DRAKE_VIEWER_DRAW` to only include updates for the model instances that have moved since the previous update.

liangfok, 2016-12-02:
@patmarion, does the above make sense?

patmarion, 2016-12-02:
By the way, the draw update message already has support for updating only individual models, or only individual links.  It will update only the links you put in the message.

I'm not sure how `drake::systems::DrakeVisualizer` can be made aware of which models or links were modified since the last publish.

> update the current use of DRAKE_VIEWER_LOAD_ROBOT to include model instance information

I think the message already has this information, and it is reflected in the tree in the drake-visualizer app.

liangfok, 2016-12-02:
> I'm not sure how drake::systems::DrakeVisualizer can be made aware of which models or links were modified since the last publish.

`drake::systems::DrakeVisualizer`'s constructor is provided a `const` reference to a `RigidBodyTree`. If it maintains a local alias of this tree as a class member variable, it can get the `model_instance_id` information, allowing it to distinguish between model instances within the tree.

Each time `drake::systems::DrakeVisualizer::DoPublish()` is called, it is provided with a `drake::systems::Context` object, which contains the generalized coordinate state. It can maintain a local copy of this state that stores the value as of the previous publish. By comparing the current generalized state with the previous generalized state, it can detect what changed since the last publish.

> > update the current use of DRAKE_VIEWER_LOAD_ROBOT to include model instance information
> 
> I think the message already has this information, and it is reflected in the tree in the drake-visualizer app.

You're right! I just double-checked [drake_visualizer.cpp#L76](https://github.com/RobotLocomotion/drake/blob/ff6f57d78e2e3feece983fa4c508048cc45cbd2e/drake/multibody/rigid_body_plant/drake_visualizer.cc#L76) and see that it does indeed save the `model_instance_id` information. Now we just need to make better use of this `model_instance_id` information when updating the visualizer.

jwnimmer-tri, 2016-12-02:
So this is a performance improvement in terms of reducing the breadth of draw updates?  Why do we believe that the breadth of draw messages is the bottleneck is visualizing large scenes?

liangfok, 2016-12-02:
Yes, as of now, this is just a performance improvement. I don't believe it's a bottleneck given the complexity of our current simulations. Maybe it'll become a problem as we scale to more model instances. I'll label this as "priority: backlog" until we determine the breadth of the model instances is causing visualization problems.

The ability to dynamically load and unload model instances may become useful once we dynamically add and remove vehicles in our vehicle simulations. This would be cleaner than simply providing poses that move vehicles outside of the field of view when they are not participating in the current simulation state, which is what I believe we're currently doing.

rdeits, 2017-04-18:
By the way, the new drake-visualizer tree viewer interface already supports adding and removing individual links, models, or groups: https://openhumanoids.github.io/director/user_guide/tree_viewer_interface.html

liangfok, 2017-04-18:
@rdeits: Thanks! I updated this issue's title and description to reflect the need to switch to the tree viewer interface.

jwnimmer-tri, 2018-06-09:
This issue was a speculative suggestion to change the implementation based on a few different hypothetical future problems.  If we have real problems, we can open new issues.  For now, this seems like a dead end.

closed 2018-06-09 by jwnimmer-tri