RobotLocomotion/drake #4348: Treatment of non-moving ("static") geometry in Drake

SeanCurtis-TRI, 2016-12-02:
Problem
======

There are several advantages in being able to partition the geometry (collision/visual/whatever) in the simulation between those that *can* move and those that *can't*.  There are a number of issues to resolve here.

1) While non-moving geometry is typically referred to as "static" in the literature, from a code-base perspective it is a nightmare.  The word "static" is horribly overloaded.  (Just try searching through issues to determine which ones are about "static geometry".)  A Drake-specific term to help disambiguate this would help.
2) Although (or perhaps because) static geometry has been much discussed (see #4333, #2709, #2662, #2661, #2644, #2481, #2003, #1933), the state of the code is somewhat chaotic.  For example, 
   a) [this code](https://github.com/RobotLocomotion/drake/blob/master/drake/multibody/parser_sdf.cc#L289) attempts to set a collision element as static, even though the code is unreachable (discussed in #4333).
   b) [this code](https://github.com/RobotLocomotion/drake/blob/master/drake/multibody/rigid_body_tree.cc#L268) attempts to "update" static geometry but identifies them as bodies with no parent. (This isn't correct for a valid tree; the only body in a valid tree without a parent should be the root, world body.)

Why do we care?
-----------------------

There are a number of advantages in distinguishing moving and immovable elements.  

1) Immovable elements allow aggressive run-time optimizations.
2) We have more geometric representations available to us for immovable geometry,
3) Immovable geometry is most likely to be shared across multiple systems; sharing is facilitated with this classification

Proposal
------------

1) Nomenclature: We refer to geometry that *cannot* move during the duration of the simulation as being "anchored".  More formally, geometry is anchored if it cannot move relative to the world frame.
   a) This name might suggest that if something is anchored, it can be "freed", in some sense.  In the short term, the anchored state of geometry is defined at initialization and remains unchanging through the duration of the simulation.  In future versions, it can be possible to change the state of geometry to and from being "anchored."
2) Ultimately, when the so-called `Geometry World` exists, all of the `System`s which introduce geometry into the simulation will be "register" the geometry with `Geometry World`.  At the time of registration, the origin system must have already characterized if the geometry is anchored.
3) Short term, we will modify the `RigidBodyTree` to provide logic to identify which collision elements should be considered static.  This logic is based on determining that there is a path from the collision element's parent body to the world, where all joints are *fixed*.  A body will have the method: `bool RigidBody::IsRigidlyFixedToGround()` which is defined recursively.

Tagging various people who have played and thought about static geometry.  Please weigh in on anything you think is missing or has simply missed the boat.
@amcastro-tri @liangfok @sherm1 @david-german-tri

david-german-tri, 2016-12-02:
I don't have anything to add; I think this is spot-on.  Although you didn't stick to your nomenclature in (3).  ;-)

SeanCurtis-TRI, 2016-12-02:
Edited in place; sorry it makes your comment look outdated. :)

SeanCurtis-TRI, 2016-12-05:
Additional Topics
=============

SDF
------
The SDF file format supports flagging a `model` as being static.  The tag is assigned to the root `model` tag.  However, an arbitrary construction of `link`s and `joints` can be hung underneath that `model`.  The assumption is that if this is the case, all joints should be replaced (at parse time) with equivalent *fixed* joints such that the entire sub-tree would be considered *anchored*.

SeanCurtis-TRI, 2017-01-05:
Initial proposal addressed in #4554.  

Still need to resolve sdf semantics.

SeanCurtis-TRI, 2020-05-14:
Closing this as being rather out-dated.

closed 2020-05-14 by SeanCurtis-TRI