RobotLocomotion/drake #4351: Linker Error Involving RigidBodyTree<double>::transformQDotMappingToVelocityMapping

liangfok, 2016-12-03:
# The Error

Using OSX 10.11.6, `clang` version Apple LLVM version 7.3.0 (clang-703.0.29), and SHA cf90c22718c3740617c9702686f28389c692f8f4:

```
FAILED: multibody/rigid_body_plant/libdrakeRigidBodyPlant.dylib
: && /Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/c++  -Werror=all -Werror=ignored-qualifiers -Werror=overloaded-virtual -Werror=shadow -Werror=inconsistent-missing-override -Werror=sign-compare -DGTEST_DONT_DEFINE_FAIL=1 -DGTEST_DONT_DEFINE_SUCCEED=1 -DGTEST_DONT_DEFINE_TEST=1 -O2 -g -DNDEBUG  -dynamiclib -Wl,-headerpad_max_install_names -Wl,-undefined -Wl,error -o multibody/rigid_body_plant/libdrakeRigidBodyPlant.dylib -install_name @rpath/libdrakeRigidBodyPlant.dylib multibody/rigid_body_plant/CMakeFiles/drakeRigidBodyPlant.dir/drake_visualizer.cc.o multibody/rigid_body_plant/CMakeFiles/drakeRigidBodyPlant.dir/viewer_draw_translator.cc.o multibody/rigid_body_plant/CMakeFiles/drakeRigidBodyPlant.dir/contact_detail.cc.o multibody/rigid_body_plant/CMakeFiles/drakeRigidBodyPlant.dir/contact_force.cc.o multibody/rigid_body_plant/CMakeFiles/drakeRigidBodyPlant.dir/contact_info.cc.o multibody/rigid_body_plant/CMakeFiles/drakeRigidBodyPlant.dir/contact_resultant_force_calculator.cc.o multibody/rigid_body_plant/CMakeFiles/drakeRigidBodyPlant.dir/contact_results.cc.o multibody/rigid_body_plant/CMakeFiles/drakeRigidBodyPlant.dir/kinematics_results.cc.o multibody/rigid_body_plant/CMakeFiles/drakeRigidBodyPlant.dir/point_contact_detail.cc.o multibody/rigid_body_plant/CMakeFiles/drakeRigidBodyPlant.dir/rigid_body_plant.cc.o  systems/lcm/libdrakeLCMSystem2.dylib systems/framework/libdrakeSystemFramework.dylib multibody/libdrakeRBM.dylib solvers/libdrakeOptimization.dylib common/trajectories/libdrakeTrajectories.dylib lcm/libdrakeLcm.dylib /Users/liang/dev/drake-distro-1/build/install/lib/liblcm.1.3.1.dylib multibody/collision/libdrakeCollision.dylib /Users/liang/dev/drake-distro-1/build/install/lib/libBulletDynamics.dylib /Users/liang/dev/drake-distro-1/build/install/lib/libBulletCollision.dylib /Users/liang/dev/drake-distro-1/build/install/lib/libLinearMath.dylib /Users/liang/dev/drake-distro-1/build/install/lib/libBulletSoftBody.dylib multibody/shapes/libdrakeShapes.dylib multibody/joints/libdrakeJoints.dylib util/libdrakeGeometryUtil.dylib math/libdrakeMath.dylib multibody/libdrakeXMLUtil.dylib thirdParty/zlib/tinyxml2/libtinyxml2.dylib thirdParty/bsd/spruce/src/libspruce.dylib common/libdrakeCommon.dylib /Users/liang/dev/drake-distro-1/build/install/lib/libgflags.2.2.0.dylib -L/Users/liang/dev/drake-distro-1/build/install/lib -L/usr/local/Cellar/gcc/6.1.0/lib/gcc/6/gcc/x86_64-apple-darwin15.5.0/6.1.0 -L/usr/local/Cellar/gcc/6.1.0/lib/gcc/6/gcc/x86_64-apple-darwin15.5.0/6.1.0/../../.. -lipopt -ldl -lcoinmumps -lgfortran -lSystem -lquadmath -lcoinmetis -lcoinlapack -lcoinblas -lgfortran -lSystem -lquadmath -lcoinmetis -lcoinlapack -lcoinblas /Users/liang/dev/drake-distro-1/build/install/lib/libnlopt.0.9.0.dylib -lm -Wl,-rpath,/Users/liang/dev/drake-distro-1/build/drake/systems/lcm -Wl,-rpath,/Users/liang/dev/drake-distro-1/build/drake/systems/framework -Wl,-rpath,/Users/liang/dev/drake-distro-1/build/drake/multibody -Wl,-rpath,/Users/liang/dev/drake-distro-1/build/drake/solvers -Wl,-rpath,/Users/liang/dev/drake-distro-1/build/drake/common/trajectories -Wl,-rpath,/Users/liang/dev/drake-distro-1/build/drake/lcm -Wl,-rpath,/Users/liang/dev/drake-distro-1/build/install/lib -Wl,-rpath,/Users/liang/dev/drake-distro-1/build/drake/multibody/collision -Wl,-rpath,/Users/liang/dev/drake-distro-1/build/drake/multibody/shapes -Wl,-rpath,/Users/liang/dev/drake-distro-1/build/drake/multibody/joints -Wl,-rpath,/Users/liang/dev/drake-distro-1/build/drake/util -Wl,-rpath,/Users/liang/dev/drake-distro-1/build/drake/math -Wl,-rpath,/Users/liang/dev/drake-distro-1/build/drake/thirdParty/zlib/tinyxml2 -Wl,-rpath,/Users/liang/dev/drake-distro-1/build/drake/thirdParty/bsd/spruce/src -Wl,-rpath,/Users/liang/dev/drake-distro-1/build/drake/common && :
Undefined symbols for architecture x86_64:
  "Eigen::Matrix<Eigen::CwiseNullaryOp<Eigen::internal::scalar_identity_op<double>, Eigen::Matrix<double, -1, -1, 0, -1, -1> >::Scalar, -1, -1, 0, -1, -1> RigidBodyTree<double>::transformQDotMappingToVelocityMapping<Eigen::CwiseNullaryOp<Eigen::internal::scalar_identity_op<double>, Eigen::Matrix<double, -1, -1, 0, -1, -1> > >(KinematicsCache<Eigen::CwiseNullaryOp<Eigen::internal::scalar_identity_op<double>, Eigen::Matrix<double, -1, -1, 0, -1, -1> >::Scalar> const&, Eigen::MatrixBase<Eigen::CwiseNullaryOp<Eigen::internal::scalar_identity_op<double>, Eigen::Matrix<double, -1, -1, 0, -1, -1> > > const&)", referenced from:
      drake::systems::RigidBodyPlant<double>::EvalTimeDerivatives(drake::systems::Context<double> const&, drake::systems::ContinuousState<double>*) const in rigid_body_plant.cc.o
ld: symbol(s) not found for architecture x86_64
clang: error: linker command failed with exit code 1 (use -v to see invocation)
```

liangfok, 2016-12-03:
This error is not reproducible on an Ubuntu 16.04 machine.

RussTedrake, 2016-12-04:
i'm seeing it, too.  as far as I'm concerned, master is broken.  we need to figure out why the build servers aren't catching it.  
```
drake008% clang --version
Apple LLVM version 8.0.0 (clang-800.0.38)
Target: x86_64-apple-darwin15.6.0
```

liangfok, 2016-12-04:
Looks like the build servers _did_ catch the error, but that it started to occur after everyone left the office on Friday (first error detected around 6:45pm Friday evening):

https://drake-jenkins.csail.mit.edu/view/Continuous%20Production/

liangfok, 2016-12-04:
I wonder if OSX should be made pre-merge again? Here's why we disabled it originally: #2793.

closed 2016-12-04 by RussTedrake

david-german-tri, 2016-12-05:
> Looks like the build servers did catch the error, but that it started to occur after everyone left the office on Friday (first error detected around 5:50pm EST Friday evening):

Yep.  For the archives, the failure appears here, and is fixed in the next build: https://drake-jenkins.csail.mit.edu/view/Continuous%20Production/job/mac-clang-continuous-release/1181/

>  I wonder if OSX should be made pre-merge again?

I'd argue against.  Episodes like this one are rare, and the OS X builders were unreliable in part because pre-merge builds were overloading them.

RussTedrake, 2016-12-06:
plus, the mac developers are powerful and able to resolve strange errors in a single bound (even on a friday night).

PR #4356 (2016-12-04): fix linker errors on mac clang reported in #4351
@liangfok for both reviews, please.  
fyi - @amcastro-tri 

resolves #4351

---
This change is [[image]](https://reviewable.io/reviews/robotlocomotion/drake/4356)