RobotLocomotion/drake #4352: refactor CreateDefaultContext

RussTedrake, 2016-12-03:
Currently `System<T>::CreateDefaultContext()` does a few things:
- Allocates memory (which also defines the derived types of the context elements, e.g. AcrobotState)
- Sets the state to the default values.
- Sets the parameters to the default values.  

For many applications, I think we need to be able to call these individually.  For instance, I need to be able to get the default state values without reallocating memory.  I propose changing the API to three methods:  `System<T>::AllocateContext()`, `System<T>::SetDefaultState(Context*)`, `System<T>::SetDefaultParameters(Context*)`.  I'm ok leaving `CreateDefaultContext()` as a convenience method, but do not want it to be virtual.  Derived classes must implement the individual methods, or use the base class defaults (e.g. states all zero).

Would love any feedback (@david-german-tri , @sherm1 , @jwnimmer-tri, et al ).  Otherwise I will PR this next time I have a few minutes to code.

RussTedrake, 2016-12-03:
fwiw - I'm also planning to add `System<T>::SetRandomState(Context*, std::RandomNumberEngine*)` which, by default, sets the default state + some Gaussian random numbers, and `System<T>::SetRandomParameters(Context*, std::RandomNumberEngine*)` which, by default, sets the default parameters + some Uniform random numbers.

jwnimmer-tri, 2016-12-03:
My understanding is that Alllocate is only _supposed_ to be the first part (reserving memory and subtypes), and something like #4191 is supposed to establish the defaults.  PRs have been defected that they rely on "CreateDefaultContext"'s established values to set up `t=0` values for simulation.

RussTedrake, 2016-12-03:
@jwnimmer-tri - Agreed on Allocate.  But the term CreateDefault implies more.  It sounds like you support the propsal.

sherm1, 2016-12-03:
I think it is appropriate that the default context contain user-specified default values for state & parameters.  That allows for immediate simulation with a new Context and makes for nice examples. In the RAII spirit the Allocate methods do need to set *some* defaults though. If we have separate SetDefault methods, the Allocate methods could end by invoking the corresponding SetDefault.

But if you think of the "create default context" operation as really "acquire needed context resources" we either need a lot of independent virtuals or a catch-all `DoCreateDefaultContext()` virtual where a leaf system can get everything it needs (e.g. cache entries) and register dependencies for proper invalidation. I prefer having the virtual since it allows for easier addition of new resource-allocating methods without more overrides.

RussTedrake, 2016-12-03:
I think having the default values typed in twice is incorrect.  So allocatecontext should only (need to) allocate.  And because those default values need to be obtainable without allocation, i think we have to pull them out.  

Allowing people the possibility of setting default values in a override on creatdefaultcontext would only lead to misuse.

sherm1, 2016-12-03:
> I think having the default values typed in twice is incorrect. 

Me too. I wasn't proposing that. After allocation there will be numbers occupying the allocated memory. I don't think they should be garbage but they could be NaNs or zeroes. I was suggesting that since you propose a method for computing the default values, the allocation methods could call that method when they are finished (automatically, not at user discretion). 

> And because the default values need to be obtainable without allocation ...

That doesn't appear to match your proposed `SetDefaultState(Context*)` since I assume it would need memory already allocated in Context in which to write the states. It is also likely that I am misunderstanding your proposal altogether. I'll stop speculating and wait for a PR to look at.

PR #4357 (2016-12-06): proposed refactoring of CreateDefaultContext
into AllocateContext, SetDefaultState, and SetDefaultParameters

resolves #4352.

---
This change is [[image]](https://reviewable.io/reviews/robotlocomotion/drake/4357)
commit 4f64ccc6 (2016-12-06): address code review.  resolves #4191.  resolves #4352.
commit c5ae9fda (2016-12-06): proposed refactoring of CreateDefaultContext into AllocateContext, SetDefaultState, and SetDefaultParameters
resolves #4352.
closed 2016-12-06 by RussTedrake